fablab-luenen/McLighting #1: no matching function for call to WS2812FX

SirJMD, 2019-02-23:
**Describe the bug**
Imported into VSC with PlatformIO it reports two errors right away:

{
	"resource": "[path]/request_handlers.h",
	"owner": "cpp",
	"severity": 8,
	"message": "no matching function for call to 'WS2812FX::setColor(uint8_t&, uint8_t&, uint8_t&, uint8_t&)'",
	"startLineNumber": 262,
	"startColumn": 85,
	"endLineNumber": 262,
	"endColumn": 85
}

{
	"resource": "[path]/request_handlers.h",
	"owner": "cpp",
	"severity": 8,
	"message": "no matching function for call to 'WS2812FX::setColor(uint8_t&, uint8_t&, uint8_t&, uint8_t&)'",
	"startLineNumber": 1519,
	"startColumn": 91,
	"endLineNumber": 1519,
	"endColumn": 91
}

**Desktop (please complete the following information):**
VSC "about":
Version: 1.31.1 (user setup)
Commit: 1b8e8302e405050205e69b59abb3559592bb9e60
Date: 2019-02-12T02:20:54.427Z
Electron: 3.1.2
Chrome: 66.0.3359.181
Node.js: 10.2.0
V8: 6.6.346.32
OS: Windows_NT x64 6.1.7601

bpohvoodoo, 2019-02-24:
You will have to install the WS2812FX library directly from github of kitesurfer. The one in the packet manager has the same version number but is not the same for the moment.

SirJMD, 2019-02-26:
That seemed to work! Thanks!

closed 2019-02-26 by SirJMD