fablab-luenen/McLighting #4: WebSocket stops after color change (RGBW) or entering modes (RGB)

TheAndi, 2019-03-14:
First I want to thank you for that great project!

I tried to make it work with my setup (ESP8266 (NodeMCU) + SK8612 RGBW strip), but I cannot get the web interface to work properly. As soon as I try to change the color, it's not responding anymore. After I reload the page it works again. I can change effects, brightness and speed as often as I want, but a color change (no matter which channel) will only work once and then the websocket dies unfortunately.

**To Reproduce**
I flashed a fresh clone of the master branch to my ESP and then uploaded the RGBW version of the HTML file.
I also tried uploading the non RGB version of the index.html. For some reason in that case, it is the other way round (same for the toblum HTML file). I can continuously change the color on the "Wheel"-page, but as soon as I enter the "Mode"-Tab the connection dies again.

**Expected behavior**
Change to color multiple times without reloading

**ChromeLog (RGBW-HTML)**
> (index):193 Connection websockets to: ws://192.168.8.101:81
> (index):240 WebSocket Open
> (index):253 WebSocket from server: Connected
> (index):198 status {mode: 5, ws2812fx_mode: 3, ws2812fx_mode_name: "Color Wipe", speed: 157, brightness: 98, …}
> (index):219 modes (63) [{…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}]
> (index):174 Navigate to pane:  pane2
> (index):304 Mode:  1
> (index):328 Send WebSocket command: /1
> (index):253 WebSocket from server: OK
> (index):304 Mode:  2
> (index):328 Send WebSocket command: /2
> (index):253 WebSocket from server: OK
> (index):304 Mode:  3
> (index):328 Send WebSocket command: /3
> (index):253 WebSocket from server: OK
> (index):328 Send WebSocket command: ?58
> (index):253 WebSocket from server: OK
> (index):328 Send WebSocket command: %144
> (index):253 WebSocket from server: OK
> (index):344 wsSetMainColor() Set main colors to: 00FF7315
> (index):328 Send WebSocket command: #00FF7315
> (index):304 Mode:  1
> (index):328 Send WebSocket command: /1
> (index):333 ++++++++ WS call waiting, skip
> (index):304 Mode:  2
> (index):328 Send WebSocket command: /2
> (index):333 ++++++++ WS call waiting, skip
> (index):304 Mode:  3
> (index):328 Send WebSocket command: /3
> (index):333 ++++++++ WS call waiting, skip

**TerminalLog (RGBW-HTML)**

> WS: [0] Connected from 192.168.8.100 url: /
> WS: [0] get Text: /1
> WS: Set WS2812 mode: [/1]
> WS: [0] get Text: /2
> WS: Set WS2812 mode: [/2]
> WS: [0] get Text: /3
> WS: Set WS2812 mode: [/3]
> Saving cfg: {"mode":5,"ws2812fx_mode":3,"ws2812fx_mode_name":"Color Wipe","speed":157,"brightness":98,"color":[0,255,0,21,0,0,0,0,0,0,0,0]}
>  Success!
> WS: [0] get Text: ?58
> WS: Set speed to: [58]
> WS: [0] get Text: %144
> WS: Set brightness to: [144]
> Saving cfg: {"mode":5,"ws2812fx_mode":3,"ws2812fx_mode_name":"Color Wipe","speed":58,"brightness":144,"color":[0,255,0,21,0,0,0,0,0,0,0,0]}
>  Success!
> WS: [0] get Text: #00FF7315
> Set main color to: R: [255] G: [115] B: [21] W: [0]
> Saving cfg: {"mode":5,"ws2812fx_mode":3,"ws2812fx_mode_name":"Color Wipe","speed":58,"brightness":144,"color":[0,255,115,21,0,0,0,0,0,0,0,0]}
>  Success!

In the logs you can clearly see that after changing the color, the connection dies.
I also tried different browsers and my phone. Same issue... Maybe a bug in the html site?

I would really appreciate any help to make it work :)

If you need more data please let me know.

bpohvoodoo, 2019-03-17:
You have to use: 
https://github.com/FabLab-Luenen/McLightingHTML
as UI

TheAndi, 2019-03-17:
Thanks a lot!
Its working with the McLightingHTML version. But is there a way to set the white channel?

What's meant by integrated UI? No need to upload an index.htm?

Thanks in advance

bpohvoodoo, 2019-03-17:
Yes, you got it. You don't need to upload any html files if
you #define USE_HTML_MIN_GZ  in definitions, which is default.
You should see the white channel in the UI, if you select an RGB Order containing "W", for example GRBW. You can select it from the Settings in the UI, (gear symbol).

TheAndi, 2019-03-18:
Hey I think there is still some bug in the web interface. 
When I leave the page open for a bit (circle and modes still work perfectly) and change anything on the slider, the hole page freezes. Sometimes it recovers after a while but the sliders jump back and I can still not change them. After page refresh everything (including the sliders) works again for a while.

**Chrome Log**

> > RWS> send #002665FF
> (index):2215 WS send:  #00000000
> (index):1256 RWS> send #00000000
> (index):2215 WS send:  #002BFF51
> (index):1256 RWS> send #002BFF51
> (index):2215 WS send:  #00218DFF
> (index):1256 RWS> send #00218DFF
> (index):2215 WS send:  #00734AFF
> (index):1256 RWS> send #00734AFF
> (index):2215 WS send:  #00FF3727
> (index):1256 RWS> send #00FF3727
> (index):2215 WS send:  #00FCFF30
> (index):1256 RWS> send #00FCFF30
> (index):2215 WS send:  $
> (index):1256 RWS> send $
> (index):1256 RWS> message event
> (index):2111 WebSocket from server: OK
> (index):1256 RWS> message event
> (index):2111 WebSocket from server: {"mode":5,"ws2812fx_mode":0,"ws2812fx_mode_name":"Static","speed":135,"brightness":255,"color":[0,252,255,48,0,0,0,0,0,0,0,0]}
> 122[Violation] 'input' handler took <N>ms
> (index):1504 [Violation] 'input' handler took 169ms
> (index):1476 [Violation] 'input' handler took 208ms
> (index):1476 [Violation] 'input' handler took 242ms
> (index):1408 [Violation] 'input' handler took 169ms
> (index):1476 [Violation] 'input' handler took 303ms
> (index):1408 [Violation] 'input' handler took 195ms
> (index):1476 [Violation] 'input' handler took 324ms
> contentscript.js:1147 [Violation] 'input' handler took 253ms
> (index):1408 [Violation] 'input' handler took 282ms
> (index):1476 [Violation] 'input' handler took 445ms
> (index):1408 [Violation] 'input' handler took 375ms
> (index):1476 [Violation] 'input' handler took 528ms
> (index):1408 [Violation] 'input' handler took 370ms
> (index):1476 [Violation] 'input' handler took 512ms
> (index):1408 [Violation] 'input' handler took 409ms
> (index):1476 [Violation] 'input' handler took 556ms
> (index):1408 [Violation] 'input' handler took 451ms
> (index):1476 [Violation] 'input' handler took 590ms
> contentscript.js:1147 [Violation] 'input' handler took 489ms
> (index):1408 [Violation] 'input' handler took 523ms
> (index):1476 [Violation] 'input' handler took 679ms
> (index):1417 [Violation] 'input' handler took 264ms
> (index):1495 [Violation] 'input' handler took 391ms
> (index):1495 [Violation] 'input' handler took 164ms
> (index):1495 [Violation] 'input' handler took 255ms
> (index):1417 [Violation] 'input' handler took 153ms
> (index):1495 [Violation] 'input' handler took 298ms
> (index):1417 [Violation] 'input' handler took 181ms
> (index):1495 [Violation] 'input' handler took 331ms
> (index):1417 [Violation] 'input' handler took 231ms
> (index):1495 [Violation] 'input' handler took 374ms
> (index):1417 [Violation] 'input' handler took 269ms
> (index):1495 [Violation] 'input' handler took 429ms
> (index):1417 [Violation] 'input' handler took 346ms
> (index):1495 [Violation] 'input' handler took 507ms
> (index):1495 [Violation] 'input' handler took 178ms
> (index):1417 [Violation] 'input' handler took 376ms
> (index):1495 [Violation] 'input' handler took 538ms
> (index):1417 [Violation] 'input' handler took 411ms
> (index):1495 [Violation] 'input' handler took 589ms
> (index):1417 [Violation] 'input' handler took 426ms
> (index):1495 [Violation] 'input' handler took 591ms
> (index):1417 [Violation] 'input' handler took 488ms
> (index):1495 [Violation] 'input' handler took 650ms
> (index):1417 [Violation] 'input' handler took 563ms
[... 114 lines cut ...]
> (index):2215 WS send:  $
> (index):1256 RWS> send $
> (index):2215 WS send:  #00FCFF30
> (index):1256 RWS> send #00FCFF30
> (index):1256 RWS> message event
> (index):2111 WebSocket from server: OK
> (index):1256 RWS> message event
> (index):2111 WebSocket from server: {"mode":5,"ws2812fx_mode":0,"ws2812fx_mode_name":"Static","speed":135,"brightness":216,"color":[0,252,255,48,0,0,0,0,0,0,0,0]}
> (index):2215 WS send:  #0036FF5D
> (index):1256 RWS> send #0036FF5D
> (index):2215 WS send:  #0044FFEF
> (index):1256 RWS> send #0044FFEF
> (index):2215 WS send:  $
> (index):1256 RWS> send $
> (index):1256 RWS> message event
> (index):2111 WebSocket from server: OK
> (index):1256 RWS> message event
> (index):2111 WebSocket from server: {"mode":5,"ws2812fx_mode":0,"ws2812fx_mode_name":"Static","speed":135,"brightness":216,"color":[0,68,255,239,0,0,0,0,0,0,0,0]}

(Chrome grouped the violations so most of them should be at the end)
Once it even crashed the tab.

Whenever this occurs you can see the violations in the log. The log on the ESP does only show something when the command actually works.

Maybe this could be fixed before the merge.

bpohvoodoo, 2019-03-18:
I'm not able to reproduce this behavior. No problems here so far.

TheAndi, 2019-03-18:
I made a video for you, if you want to take a look:
https://www.file-upload.net/download-13542504/2019-03-18_23h56_02.mp4.html

As soon as I touch the sliders it gets laggy. But as soon as I reload, everything is fine.

Chrome Version 72.0.3626.121 (Official Build) (64-Bit)
But I remember it being the same for other browsers.

bpohvoodoo, 2019-03-19:
You can try again, I think i found it.

TheAndi, 2019-03-19:
Yes you did! Thanks a lot. 
I had the page open since 3h to test and it still works.

One more question: is there a reason why changes in color and the sliders to "so long" to apply (about 1-2secs). Because when I change effects this happens almost instantly (can be tested very well with hitting the strobe several times) and the sliders/colors don't.

Sorry I am a bit spoiled by the hyperion project on my TV :D (which is basically instant)

bpohvoodoo, 2019-03-20:
That is normal and a wanted behavior.

closed 2019-03-20 by bpohvoodoo

TheAndi, 2019-03-20:
> That is normal and a wanted behavior.

So you want to say it's on purpose, that there is lag between selecting a color and seeing it?
Please don't get me wrong I really enjoy that project, but shouldn't it be as responsive as possible?

(I was just wondering because the effects do apply really responsive and wanted to know the reason)

bpohvoodoo, 2019-03-21:
I made it faster, as requested, reduced from 500ms to 50ms. But the last 50 ms delay i need to prevent flooding.

VeniceNerd, 2019-07-10:
Have these two versions ever been merged or do I still need to use @bpohvoodoo 's version to get W channel support?

Thanks! :)

bpohvoodoo, 2019-07-30:
The Versions have not been merged so, far. I merge every change from the original project into mine.