fablab-luenen/McLighting #6: NUMLEDS > 466 LEDs

MarcoA12000, 2019-03-17:
Tolle Arbeit! 
Jedoch ist in der aktuellen Version eine Änderung der NUMLEDS auf über 466 LEDs ist nicht erfolgreich.

bpohvoodoo, 2019-03-17:
Inwiefern. Programmtechnisch sollten 65535 gehen. Allerdings reicht hierfür der Speicher nicht.
Jede LED braucht 4 Byte Speicher. Also sollten Speichertechnisch ca. 1000 gehen.

MarcoA12000, 2019-03-17:
Hi!
Ich habe hier zwei Streifen mit je 300 LEDs in Reihe hintereinander (=600 LEDs). Nach dem ersten Start wurden nur ca. 88 LEDs angesprochen, obwohl in der definitions.h "#define NUMLEDS 600" hinterlegt war und beim Einstellen des WLANs über 192.168.4.1 auch die 600 LEDs angegeben wurden. Beim Setup in der Weboberfläche (Zahnrad unten rechts) kann ich die Anzahl wieder ändern - jedoch nur max. bis 466 - bei höheren Werten werden nur noch die ersten paar LEDs angesprochen...

bpohvoodoo, 2019-03-18:
So habe ich auf 700 LEDs aufgebohrt, da ich noch den Speicher noch optimieren konnte. Mehr ist (zumindest im Moment nicht drin)!
Teste mal den Development Branch, ob es auch tatsächlich funktioniert. Zumindest crasht er nicht mehr ;-)

MarcoA12000, 2019-03-19:
Hi!
Also, ich habe alles aus dem Development Branch herunter geladen und neu installiert. Doch der Fehler bleibt bestehen. Allerdings können jetzt max. 544 LEDs angesprochen werden. 
Jedoch ist mir aufgefallen, dass je nach laufendem Programm mal mehr oder weniger LEDs bei einer Änderung angenommen werden. Am stabilsten ließ sich eine Änderung im STATIC Programm ausführen.
Mir ist aufgefallen, dass bei einer Änderung auf bspw. 600 LEDs beim Speichern alle 600 LEDs kurz aufleuchten - kurz danach jedoch startet das aktuelle Programm neu und steuert nur noch einen kleinen Teil am Anfang an, der restliche Teil der LEDs bleibt dann hell erleuchtet.... In einigen Programmen ist eine Änderung nur nach einigen Speichervorgängen erfolgreich (vorher kommt jeweils eine Fehlermeldung, man möchte es noch einmal speichern...)
Deaktivieren von E131, TV und/oder GY33 bringen keine Änderung.
Beim neuen Einspielen des Sketches ist die Anzahl der LEDs in der Weboberfläche jedes mal unterschiedlich, mal 33, 44 und 88 konnte ich beobachten.
Ich hoffe, dir ein wenig helfen zu können (Die vorherige master-Version vom 28.02.2019 lief ja ohne Probleme)
Grüße...
MarcoA

bpohvoodoo, 2019-03-19:
Hmm, habe im Moment keine Ahnung woran das liegt.
Hast du mit Flash Size 4MB (2MB SPIFFS), CPU 160 mHz und IwIP Higher Bandwith compiliert?

bpohvoodoo, 2019-03-19:
Kommentiere mal #define CUSTOM_WS2812FX_ANIMATIONS aus. Vielleicht liegt es an der Fastled Lib

bpohvoodoo, 2019-03-21:
So das liegt an DMA der braucht einfach viel Speicher. Wenn du auf DMA verzichtest, dann gehen deutlich mehr LED.
also //define USE_WS2812FX_DMA auskommentieren

closed 2019-03-21 by bpohvoodoo