fablab-luenen/McLighting #8: #define ENABLE_MQTT 1 causes program not to work

TheAndi, 2019-03-20:
When I uncomment `//#define ENABLE_MQTT 1 ` the ESP is not starting up correctly and just produces lots of TerminalLog. 

As I do not use MQTT, I do not need it to work. I just wanted to test the code and found that. 
It might be related to my setup or something I did wrong. So if it usually works, consider this low priority.

I also tried `//#define ENABLE_MQTT 0 ` which leads to other problems:
`McLighting.ino:77:30: fatal error: PubSubClient.h: No such file or directory`
I thought PIO installs all the dependencies? So I installed it manually. 
This leads to 
`Arduino\McLighting\request_handlers.h:943:50: error: 'measureJson' was not declared in this scope`  and others.

**To Reproduce**
Clean install of master or develop branch (using PIO)

**Expected behavior**
Should not run into issues

**TerminalLog** (with #define ENABLE_MQTT 1)

```
Starting...Main Setup
Enabled Button Mode on PIN: 14
FS File: /index.htm, size: 96.50KB
FS File: /config.json, size: 84B
FS File: /uistate.json, size: 203B
FS File: /stripstate.json, size: 128B
FS Usage: 101404/957314 bytes
Reading config file... Opened!
Config: Parsed!
{"hostname":"McLightingRGBW","ws_cnt":82,"ws_rgbo":"GRBW","ws_pin":3,"ws_fxopt":120}

Exception (28):
epc1=0x40230e9a epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont
sp: 3ffffb50 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffcf0:  3fff22a4 00000000 3fff20b0 40229388
3ffffd00:  3fff22a4 00000054 3fff20b0 4021a0fd
3ffffd10:  3fff44d0 3fff44d0 3ffffd30 4021a0fd
3ffffd20:  3ffeb498 00000054 3fff20b0 00000001
3ffffd30:  3fff22a4 00000054 3ffffd60 4020c318
3ffffd40:  00000000 00000054 3fff20b0 4020ff00
3ffffd50:  3fff42dc 3fff42dc 3fff44d0 3fff4520
3ffffd60:  3fff4510 3fff44d0 3fff38a0 4022f2c0
3ffffd70:  00000000 000003e8 0000020a 00000000
3ffffd80:  00000000 3ffffeb0 3fff3834 3ffffd60
3ffffd90:  3ffffd50 3fff38e4 3ffffe10 00000000
3ffffda0:  3fff38c4 000e9b82 3fff38c4 4022b317
3ffffdb0:  3fff22a4 3ffffeb0 3fff3834 0000008c
3ffffdc0:  3fff22a4 3ffffeb0 3fff20b0 40211ba3
3ffffdd0:  feefeffe feefeffe feefeffe feefeffe
3ffffde0:  feefeffe feefeffe feefeffe feefeffe
3ffffdf0:  feefeffe feefeffe feefeffe feefeffe
3ffffe00:  feefeffe feefeffe feefeffe feefeffe
3ffffe10:  000e9b82 00018c1c 00002000 00000100
3ffffe20:  00000005 00000020 feefeffe feefeffe
3ffffe30:  feefeffe feefeffe feefeffe feefeffe
3ffffe40:  feefeffe feefeffe feefeffe feefeffe
3ffffe50:  feefeffe feefeffe feefeffe feefeffe
3ffffe60:  feefeffe feefeffe feefeffe feefeffe
3ffffe70:  feefeffe feefeffe feefeffe feefeffe
3ffffe80:  feefeffe feefeffe feefeffe feefeffe
3ffffe90:  feefeffe feefeffe feefeffe 00000000
3ffffea0:  00000000 00000000 feefeffe feefeffe
3ffffeb0:  3fff38c4 3fff3834 feefeffe feefeffe
3ffffec0:  feefeffe feefeffe feefeffe feefeffe
3ffffed0:  feefeffe feefeffe feefeffe feefeffe
3ffffee0:  feefeffe feefeffe feefeffe feefeffe
3ffffef0:  feefeffe feefeffe feefeffe feefeffe
3fffff00:  feefeffe feefeffe feefeffe feefeffe
3fffff10:  feefeffe feefeffe feefeffe feefeffe
3fffff20:  feefeffe feefeffe feefeffe feefeffe
3fffff30:  feefeffe feefeffe feefeffe feefeffe
3fffff40:  feefeffe feefeffe feefeffe feefeffe
3fffff50:  feefeffe feefeffe feefeffe feefeffe
3fffff60:  feefeffe feefeffe feefeffe feefeffe
3fffff70:  feefeffe feefeffe feefeffe feefeffe
3fffff80:  3fff3744 3ffffe9c feefeffe feefeffe
3fffff90:  feefeffe feefeffe feefeffe feefeffe
3fffffa0:  feefeffe feefeffe feefeffe 3fff2170
3fffffb0:  3fffdad0 00000000 3fff216a 4022afec
3fffffc0:  feefeffe feefeffe 3ffe88e8 40100721
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

Starting...Main Setup
[...]
```

bpohvoodoo, 2019-03-20:
Try again, but be shure to erase complete flash content.

debsahu, 2019-03-20:
@TheAndi Add `PubSubClient` to `lib_deps =` in **platformio.ini**

bpohvoodoo, 2019-03-21:
Tried every combination.  It seems to work.

closed 2019-03-21 by bpohvoodoo