fablab-luenen/McLighting #20: Strip GPIO Pin can't change

Armageddit, 2019-04-24:
**Describe the bug**
try to change the GPIO Pin

**To Reproduce**
Steps to reproduce the behavior:
change the port in definitions.h
and on the webpage in settings isnt possible

i have an ESP01S and use SK6812 RGBW strip
the board i use need GPIO2 (same pin as onboardLED)

**Desktop (please complete the following information):**
 - OS: [WIN10]
 - Browser [Firefox,Chrome,Edge]
 - Version [aktuell]

**Smartphone (please complete the following information):**
 - Device: [S9+]
 - OS: [android 9]
 - Browser [Opera]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

Armageddit, 2019-04-25:
Seriell monitor

> 
> WS: [0] get Text: Csp2
> WS: Get status info: {"hostname":"McLightingRGBW","ws_cnt":3,"ws_rgbo":"GRBW","ws_pin":3,"ws_fxopt":48}
> WS: [0] get Text: C
> WS: Get status info: {"hostname":"McLightingRGBW","ws_cnt":3,"ws_rgbo":"GRBW","ws_pin":3,"ws_fxopt":48}
> WS: [0] get Text: $
> WS: Get status info: {"mode":0,"ws2812fx_mode":0,"ws2812fx_mode_name":"Static","speed":196,"brightness":196,"color":[0,255,0,0,0,0,0,0,0,0,0,0]}

`#define ENABLE_STATE_SAVE 0           // If defined, load saved state on reboot and save state. If set to 0 from EEPROM, if set to 1 from SPIFFS`
will not help but i dont get errormessage by saving

Armageddit, 2019-04-25:
ups

Armageddit, 2019-04-26:
https://github.com/Armageddit/McLighting

closed 2019-04-26 by Armageddit

bpohvoodoo, 2019-04-28:
If you have chosen to use DMA the pin is ignored.