fablab-luenen/McLighting #21: Build problem ArduinoJson

devsys, 2019-04-26:
Hello, I saw you have added rgbw support so I thought I'd test it out. I am building using VS code and platformIO. I just successfully built and uploaded the current McLighting release before trying this one, but I'm getting this error from the verbose build output. Any hints appreciated:

> Executing task: C:\Users\Admin\.platformio\penv\Scripts\platformio.exe run --verbose --environment esp01_1m <

Processing esp01_1m (platform: espressif8266@2.0.4; build_flags: -DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH, -DMQTT_MAX_PACKET_SIZE=512, -w; lib_deps: WiFiManager@0.14, AsyncMqttClient, ArduinoJson@6.9.1, WS2812FX, NeoPixelBus, WebSockets, ESPAsyncE131, ESPAsyncUDP, https://github.com/FabLab-Luenen/GY33_MCU, Brzo I2C, IRremoteESP8266@2.5.3, FastLED; monitor_speed: 115200; framework: arduino; upload_speed: 115200; board: esp01_1m; board_build.flash_mode: dout; upload_resetmethod: nodemcu)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Storage: c:\Users\Admin\Documents\McLighting-FabLab-RGBW\.piolibdeps
Looking for WiFiManager library in registry
Found: https://platformio.org/lib/show/567/WifiManager
LibraryManager: Installing id=567 @ 0.14
WifiManager @ 0.14 is already installed
LibraryManager: Installing id=346
AsyncMqttClient @ 0.8.2 is already installed
Installing dependencies
LibraryManager: Installing id=305 @ ^1.1.0
ESPAsyncTCP @ 1.2.0 is already installed
LibraryManager: Installing id=1826 @ ^1.0.0
AsyncTCP @ 1.0.3 is already installed
LibraryManager: Installing id=64 @ 6.9.1
ArduinoJson @ 6.9.1 is already installed
LibraryManager: Installing id=1647
WS2812FX @ 1.1.9 is already installed
Installing dependencies
LibraryManager: Installing id=28 @ ~1.1.4
Adafruit NeoPixel @ 1.1.8 is already installed
LibraryManager: Installing id=547
NeoPixelBus @ 2.4.4 is already installed
LibraryManager: Installing id=549
WebSockets @ 2.1.3 is already installed
LibraryManager: Installing id=1781
ESPAsyncE131 @ 1.0.1 is already installed
Installing dependencies
LibraryManager: Installing id=359
ESPAsyncUDP @ b592ac6e93 is already installed
LibraryManager: Installing GY33_MCU
GY33_MCU @ 112c4dd is already installed
Installing dependencies
LibraryManager: Installing id=335 @ ~1.3.2
Brzo I2C @ 1.3.2 is already installed
LibraryManager: Installing id=335
Brzo I2C @ 1.3.2 is already installed
LibraryManager: Installing id=1089 @ 2.5.3
IRremoteESP8266 @ 2.5.3 is already installed
LibraryManager: Installing id=126
FastLED @ 3.2.6 is already installed
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz 80KB RAM (1MB Flash)
xtensa-lx106-elf-g++ -o "C:\Users\Admin\Documents\McLighting-FabLab-RGBW\Arduino\McLighting\McLighting.ino.cpp" -x c++ -fpreprocessed -dD -E "c:\users\admin\appdata\local\temp\tmp4sjwxv"
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Platform incompatible library c:\Users\Admin\Documents\McLighting-FabLab-RGBW\.piolibdeps\AsyncTCP_ID1826
Collected 40 compatible librariesMore details about "Library Compatibility Mode": https://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode

Scanning dependencies...
Skip platform incompatible dependency {u'platforms': [u'espressif32'], u'version': u'^1.0.0', u'name': u'AsyncTCP'}
Dependency Graph
|-- <AsyncMqttClient> 0.8.2 (c:\Users\Admin\Documents\McLighting-FabLab-RGBW\.piolibdeps\AsyncMqttClient_ID346)
|   |-- <ESPAsyncTCP> 1.2.0 (c:\Users\Admin\Documents\McLighting-FabLab-RGBW\.piolibdeps\ESPAsyncTCP_ID305)
|   |   |-- <ESP8266WiFi> 1.0 (C:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi)
|-- <ArduinoJson> 6.9.1 (c:\Users\Admin\Documents\McLighting-FabLab-RGBW\.piolibdeps\ArduinoJson_ID64)
|-- <WS2812FX> 1.1.9 (c:\Users\Admin\Documents\McLighting-FabLab-RGBW\.piolibdeps\WS2812FX_ID1647)
|   |-- <Adafruit NeoPixel> 1.1.8 (c:\Users\Admin\Documents\McLighting-FabLab-RGBW\.piolibdeps\Adafruit NeoPixel_ID28)
|-- <NeoPixelBus> 2.4.4 (c:\Users\Admin\Documents\McLighting-FabLab-RGBW\.piolibdeps\NeoPixelBus_ID547)
|   |-- <SPI> 1.0 (C:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\libraries\SPI)
|-- <WebSockets> 2.1.3 (c:\Users\Admin\Documents\McLighting-FabLab-RGBW\.piolibdeps\WebSockets_ID549)
|   |-- <ESPAsyncTCP> 1.2.0 (c:\Users\Admin\Documents\McLighting-FabLab-RGBW\.piolibdeps\ESPAsyncTCP_ID305)
|   |   |-- <ESP8266WiFi> 1.0 (C:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi)
|   |-- <Ethernet(esp8266)> 1.0.4 (C:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\libraries\Ethernet)
|   |   |-- <SPI> 1.0 (C:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\libraries\SPI)
[... 57 lines cut ...]
xtensa-lx106-elf-g++ -o .pioenvs\esp01_1m\lib5ac\ESP8266WiFi\ESP8266WiFiAP.cpp.o -c -fno-rtti -std=c++11 -w -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -fno-exceptions -Wall -DPLATFORMIO=30607 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP01 -DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH -DMQTT_MAX_PACKET_SIZE=512 -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO=10805 -DARDUINO_BOARD=\"PLATFORMIO_ESP01_1M\" -DFLASHMODE_DOUT -DLWIP_OPEN_SRC -DTCP_MSS=1460 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DVTABLES_IN_FLASH -IC:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src -IC:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\tools\sdk\include -IC:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include -IC:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\cores\esp8266 -IC:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip2\include -IC:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\variants\generic C:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp
xtensa-lx106-elf-g++ -o .pioenvs\esp01_1m\lib5ac\ESP8266WiFi\ESP8266WiFiGeneric.cpp.o -c -fno-rtti -std=c++11 -w -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -fno-exceptions -Wall -DPLATFORMIO=30607 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP01 -DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH -DMQTT_MAX_PACKET_SIZE=512 -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO=10805 -DARDUINO_BOARD=\"PLATFORMIO_ESP01_1M\" -DFLASHMODE_DOUT -DLWIP_OPEN_SRC -DTCP_MSS=1460 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DVTABLES_IN_FLASH -IC:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src -IC:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\tools\sdk\include -IC:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include -IC:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\cores\esp8266 -IC:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip2\include -IC:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\variants\generic C:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src\ESP8266WiFiGeneric.cpp
In file included from Arduino\McLighting\mode_custom_ws2812fx_animations.h:10:0,
from C:/Users/Admin/Documents/McLighting-FabLab-RGBW/Arduino/McLighting/McLighting.ino:533:xtensa-lx106-elf-g++ -o .pioenvs\esp01_1m\lib5ac\ESP8266WiFi\ESP8266WiFiMulti.cpp.o -c -fno-rtti -std=c++11 -w -Os -mlongcalls
-mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -fno-exceptions -Wall -DPLATFORMIO=30607 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP01 -DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH -DMQTT_MAX_PACKET_SIZE=512 -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO=10805 -DARDUINO_BOARD=\"PLATFORMIO_ESP01_1M\" -DFLASHMODE_DOUT -DLWIP_OPEN_SRC -DTCP_MSS=1460 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DVTABLES_IN_FLASH -IC:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src -IC:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\tools\sdk\include -IC:\Users\Admin\.plat [... cut ...]
======================================================================================================= [SUMMARY] =======================================================================================================
Environment esp01_1m    [ERROR]
Environment nodemcuv2   [SKIP]
============================================================================================== [ERROR] Took 17.38 seconds ==============================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

bpohvoodoo, 2019-04-28:
Try again.

devsys, 2019-04-28:
Fixed and working great. Thank you.

closed 2019-04-28 by devsys