fablab-luenen/McLighting #25: Is it possible to create gradients?

VeniceNerd, 2019-06-17:
Im wondering if it is possible to create gradients with this repository. For example I’d love to set saturated blue on one end of the strip and have it fade to pure white on the other end of the strip. 

No animations or anything. Just a simple gradient fade. Is that doable?

bpohvoodoo, 2019-09-19:
Feature is in development and already working.

VeniceNerd, 2019-09-19:
@bpohvoodoo omg are you serious??? That is absolutely incredible!!!! Can't wait for this to ship! Will it support pure white channel as well? So fading from the pure white LED to a mix of colors on the other end?

bpohvoodoo, 2019-09-20:
Yes, it will!

VeniceNerd, 2019-09-20:
It’s super cool. Really exciting! Do you think there will be a way to invoke those “fade recipes” in HomeKit as well? Or can they be invoked by a link? I think in that case I could use Apple Shortcuts to automate them into my scenes.

VeniceNerd, 2019-09-20:
Oh and can I help in any way? I won’t be of any use with programming but I’d be happy to test or discuss ideas. 

Would also be happy to donate to the project if you accept donations.

bpohvoodoo, 2019-09-27:
If you want you can try the development branch now. Gradients is a CustomEffect (61).

bpohvoodoo, 2019-10-08:
If you want you can try the development branch (3.1.0) now. Gradients is a CustomEffect (61).
https://github.com/FabLab-Luenen/McLighting/tree/development

closed 2019-10-20 by bpohvoodoo