fablab-luenen/McLighting #26: Can't change Strip Count and GPIO Pin

VeniceNerd, 2019-08-25:
**Describe the bug**
I am loading the web interface and press the settings button to change the GPIO Pin as well as the LED count. However, the numbers always jump back to the default. See this video:
https://www.dropbox.com/s/o3hhnjukpw919cu/McLighting.mov?dl=0

**To Reproduce**
Steps to reproduce the behavior:
1. Go to web interace
2. Click on "settings"
3. Enter a different GPIO Pin number or LED count number.
4. Number jumps back to previous number.

**Expected behavior**
Number changes to entered number.

**Screenshots**
https://www.dropbox.com/s/o3hhnjukpw919cu/McLighting.mov?dl=0

**Desktop (please complete the following information):**
 - OS: Mac OS 10.14.3
 - Browser Chrome & Safari

ACKET2009, 2019-09-07:
I have issue in changing the gpio number in a scetch. I has excluded the DMA and UART as in original version, but the lights does not work
//#define USE_WS2812FX_UART 1
//#define USE_WS2812FX_DMA 1      // 0 = Used PIN is ignored & set to RX/GPIO3; 1 = Used PIN is ignored & set to D4/GPIO2; 2 = Uses PIN is ignored & set to TX/GPIO1;  Uses WS2812FX, see: https://github.com/kitesurfer1404/WS2812FX
//#if defined(USE_WS2812FX_DMA)
//#if defined(USE_WS2812FX_UART)

  #define MAXLEDS 384           // due to memory limit of esp8266 at the moment only 384 leds are supported in DMA Mode. More can crash if mqtt is used.
//#else
  #define MAXLEDS 4096
//#endif 
// Neopixel
#define LED_PIN 2          // PIN (15 / D8) where neopixel / WS2811 strip is attached; is configurable, if USE_WS2812FX_DMA is not defined. Just for the start
#define NUMLEDS 30         // Number of leds in the; is configurable just for the start 
#define RGBORDER "GRBW"    // RGBOrder; is configurable just for the start
#define FX_OPTIONS 48      // ws2812fx Options 48 = SIZE_SMALL + FADE_MEDIUM  is configurable just for the start; for WS2812FX setSegment OPTIONS, see: https://github.com/kitesurfer1404/WS2812FX/blob/master/extras/WS2812FX%20Users%20Guide.md
//#define LED_TYPE_WS2811    // Uncomment, if LED type uses 400 KHz (classic 'v1' (not v2) FLORA pixels, WS2811 drivers)
#define LED_BUILTIN 4      // ESP-12F has the built in LED on GPIO2, see https://github.com/esp8266/Arduino/issues/2192

bpohvoodoo, 2019-09-08:
All is as expected and documented in the  definitions.h

bpohvoodoo, 2019-09-08:
@ACKET2009 I will have a look.

bpohvoodoo, 2019-09-13:
Depending on which OTA you have enabed,
please try http://ipofyourdevice/update

Here you can upload a bin file.

closed 2019-09-13 by bpohvoodoo

VeniceNerd, 2019-09-13:
Forgive my ignorance, but when I download the repository I get a zip file. Am I supposed to upload that? When I unzip the file I can find any bin files inside. Sorry if this is super obvious.

bpohvoodoo, 2019-09-14:
No. You have to compile in Arduino IDE or Visual Studio Code.
The compiled bin file can be uploaded.
Here for Arduino IDE:
[image]
The bin file can be found in the sketch folder.