fablab-luenen/McLighting #27: exit status 1 no matching function for call to 'NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeed800Kbps>::Update()'

alexoxo7, 2019-09-03:
I everyone,
 thx for your wonderfull project. the bug appear when i try to upload the skecth

image below 

https://ibb.co/x1dMmBg

Arduino : 1.8.9 (Windows 10), Carte : "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (1M SPIFFS), v2 Higher Bandwidth, Disabled, None, Only Sketch, 115200"

D:\telechargement\McLighting-master\McLighting-master\Arduino\McLighting\McLighting.ino: In function 'void DMA_Show()':

McLighting:170:19: error: no matching function for call to 'NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeed800Kbps>::Update()'

       dma->Update();

                   ^

D:\telechargement\McLighting-master\McLighting-master\Arduino\McLighting\McLighting.ino:170:19: note: candidate is:

In file included from C:\Users\alex sing\Documents\Arduino\libraries\NeoPixelBus-master\src/NeoPixelBus.h:68:0,

                 from D:\telechargement\McLighting-master\McLighting-master\Arduino\McLighting\McLighting.ino:113:

C:\Users\alex sing\Documents\Arduino\libraries\NeoPixelBus-master\src/internal/NeoEsp8266DmaMethod.h:297:26: note: void NeoEsp8266DmaMethodBase<T_SPEED>::Update(bool) [with T_SPEED = NeoEsp8266DmaSpeed800Kbps]

     void ICACHE_RAM_ATTR Update(bool)

                          ^

C:\Users\alex sing\Documents\Arduino\libraries\NeoPixelBus-master\src/internal/NeoEsp8266DmaMethod.h:297:26: note:   candidate expects 1 argument, 0 provided

In file included from sketch\mode_custom_ws2812fx_animations.h:10:0,

                 from D:\telechargement\McLighting-master\McLighting-master\Arduino\McLighting\McLighting.ino:267:

C:\Users\alex sing\Documents\Arduino\libraries\FastLED-master/FastLED.h: At global scope:

C:\Users\alex sing\Documents\Arduino\libraries\FastLED-master/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.002

 #    pragma message "FastLED version 3.003.002"

                     ^

In file included from C:\Users\alex sing\Documents\Arduino\libraries\FastLED-master/FastLED.h:65:0,

                 from sketch\mode_custom_ws2812fx_animations.h:10,

                 from D:\telechargement\McLighting-master\McLighting-master\Arduino\McLighting\McLighting.ino:267:

C:\Users\alex sing\Documents\Arduino\libraries\FastLED-master/fastspi.h:130:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output

 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

                       ^

Plusieurs bibliothèque trouvées pour "Adafruit_NeoPixel.h"
Utilisé : C:\Users\alex sing\Documents\Arduino\libraries\Adafruit_NeoPixel
Non utilisé : C:\Users\alex sing\Documents\Arduino\libraries\Adafruit_NeoPixel-master
exit status 1
no matching function for call to 'NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeed800Kbps>::Update()'

Ce rapport pourrait être plus détaillé avec
l'option "Afficher les résultats détaillés de la compilation"
activée dans Fichier -> Préférences.

Gwenselah, 2019-09-06:
Hello

I have the same error.
Clean build with all actual resources from github.

What can I do to fix the issue?

kind regards

ACKET2009, 2019-09-08:
Hello guys! I have the same issues. I has only comment the string #define USE_WS2812FX_DMA 0 and the GPIO3 (RX) pin on my ESP8266 mini for the LED strip...
And it works.
But I also need the answer to the question - how to fix the problem with DMA (that You show, thank's!), if I want to use it?

Makuna, 2019-09-08:
PLEASE do not use the DMA method the way you do it. I may change the internal API and it will break you as you have found out.  Update must be called with a bool argument now.

bpohvoodoo, 2019-09-08:
@Makuna How should we use it ?

Makuna, 2019-09-08:
Don't use the method directly, use the NeoPixelBus object.  
The "features" and "method" classes are definition classes; and their methods and members can change on my whim to support new features and fix bugs.  So things like you are seeing can and will continue to happen.  While the NeoPixelBus methods are the API and will not change (barring some extreme case).

My examples show you how to do it, and here is a very "raw" way to instance it for Ws2812x (x meaning any Ws2812 with a letter following the model).

```
NeoPixelBus<NeoGrbFeature, NeoEsp8266DmaWs2812xMethod> strip(LED_COUNT);
```
then use strip in a very similar API to the Adafruit library; or access the raw buffer using `Pixels()` method.

closed 2019-09-10 by bpohvoodoo

rlesac, 2019-10-06:
I'm still getting that error when trying to compile the sketch.

bpohvoodoo, 2019-10-07:
It seems that you havn't the latest version of the neopixelbus library.

rlesac, 2019-10-07:
> 
> 
> It seems that you havn't the latest version of the neopixelbus library.

I somehow missed to update that library. Thanks!