fablab-luenen/McLighting #28: setMode From HTTP Rest API Doesn't Change Color

ryancasler, 2019-09-04:
I'm unable to use the HTTP Rest API to change the colors in an animation to what they are set to in the web UI.  The response from the board is correct but the animation uses the old colors.  This is the url that I am using:

http://192.168.1.123/set_mode?m=15&r=255&g=0&b=0&r2=0&g2=0&b2=0&c=100&s=180

And this is the response.

{"mode":7,"ws2812fx_mode":15,"ws2812fx_mode_name":"Fade","speed":180,"brightness":254,"color":[0,255,0,0,0,0,0,0,0,0,0,0]}

However, instead of fading between black and red it fades between blue and red which were the last settings I entered into the webUI.  The only thing it appears that I am able to change through the HTTP rest api is the mode.  Can you please tell me what I am doing wrong?

bpohvoodoo, 2019-09-08:
I am investigating on this issue

bpohvoodoo, 2019-09-08:
Hello use
the 'all' parameter as set_mode only changes mode on the fly. The rest of parameters should be ignored, but they arent at the moment.
I correct this and will update the Rest-API docu.
For the moment use:
http://192.168.1.123/all?m=15&r=255&g=0&b=0&r2=0&g2=0&b2=255&c=100&s=180

bpohvoodoo, 2019-09-08:
REST-API Documentation is no up to date!

bpohvoodoo, 2019-09-08:
Please test the new version.

ryancasler, 2019-09-09:
Ah!   Yes, that would make much more sense.  Well, that would mean changing a lot of the implementations I currently have using Toblum's original fork and it would mean changing all 5 boards at once (since I can't have some on the old and some on the new and still have everything work.  I'll have to do that when I have slightly more time on my hands.  But thanks for clarifying!!

ryancasler, 2019-09-09:
I can confirm that is is working using the /all instead.  Thank you!!!

ryancasler, 2019-09-09:
Scratch that...it still doesn't work.   
I use:  http://192.168.1.142/all?m=15&rgb=0000FF&rgb2=000000&s=180&c=100
and it still doesn't work.  It reports back that the correct color is being used but it still doesn't set the color correctly.   I have to issue the request to the board twice in order for it to actually make the change that I requested,  I went through and updated all my boards based on you saying this was fixed but now nothing is working.  Any help you can give to sort this out would be much appreciated.

bpohvoodoo, 2019-09-10:
I tested it and it is working.
first of all you have to give WRGB Values please add 00 in front of your color  values
as this version has an w channel, see Rest-API doku, even if you don't use rgbw strips.

rgb: Value for color as HEX WRGB value, e.g. 0004d2ff
rgb2: Value for color as HEX WRGB value, e.g. 0004d2ff.
rgb3: Value for color as HEX WRGB value, e.g. 0004d2ff

bpohvoodoo, 2019-09-10:
Maybe I found what is causing this.