fablab-luenen/McLighting #30: Make the AUTOrun by default.

ACKET2009, 2019-09-09:
Hello!
I'm trying to enable the auto play checked effects automaticaly when the Led stripe powered on, but can not do it. Can You make this option by default, or make a key in definitions file to on or off this feature  when somebody need?

ACKET2009, 2019-09-09:
Can somebody help?

ACKET2009, 2019-09-10:
Somebody know, how to use the autoplay by default?

bpohvoodoo, 2019-09-10:
It is not possible at the moment, as this status is not saved, to neither spiffs nor eeprom.
I will have to change code.
I will have a look.

ACKET2009, 2019-09-10:
Many thanks! I will awaiting the result with a good news! :^)

viktor6, 2019-09-18:
I will also wait for the result

bpohvoodoo, 2019-09-19:
It is in development. "By default" will mean that the status "auto" is beeing saved. On reboot it will restart in "auto" mode.

viktor6, 2019-09-20:
Is it possible to turn on auto mode by http request?

bpohvoodoo, 2019-09-20:
https://github.com/FabLab-Luenen/McLighting/wiki/HTTP-REST-API#get-lightmode-http10

https://ipofyourdevice/auto

viktor6, 2019-09-20:
I read it and ask
http://192.168.0.120/auto
File Not Found

URI: /auto
Method: GET
Arguments: 0

bpohvoodoo, 2019-09-21:
Ok. This seems to bee ab bug. 
Will have a look.

bpohvoodoo, 2019-09-21:
I checked it here and it is working. Which version do you use?
here is mine:
http://ipofyourdevice/esp_status
{"HOSTNAME":"McLightingRGBW",
"version":"2.2.6.RU1.rgbw.3c",
"heap":18528,
"sketch_size":509040,
"free_sketch_space":1585152,
"flash_chip_size":4194304,
"flash_chip_real_size":4194304,
"flash_chip_speed":40000000,
"sdk_version":"2.2.1(cfd48f3)",
"core_version":"2_5_2",
"cpu_freq":160,
"chip_id":1458415,
"animation_lib":"WS2812FX_DMA",
"ws2812_pin":3,
"led_count":194,
"rgb_order":"GRBW",
"rgbw_mode":"ON",
"button_mode":"ON",
"button_pin":14,
"button_gy33":"ON",
"gy33_pin":12,
"ir_remote":"OFF",
"mqtt":"AMQTT",
"home_assistant":"ON",
"legacy_animations":"ON",
"tv_animation":"ON",
"e131_animations":"ON",
"ota":"HTTP",
"state_save":"SPIFFS"}

viktor6, 2019-09-21:
I reloaded the sketch and everything became normal
{"HOSTNAME":"McLighting",
"version":"2.2.6.RU1.rgbw.3c",
"heap":20024,
"sketch_size":493920,
"free_sketch_space":2650112,
"flash_chip_size":4194304,
"flash_chip_real_size":4194304,
"flash_chip_speed":40000000,
"sdk_version":"2.2.1(cfd48f3)",
"core_version":"2_5_2",
"cpu_freq":80,
"chip_id":1458208,
"animation_lib":"WS2812FX_DMA",
"ws2812_pin":3,
"led_count":179,
"rgb_order":"GRB",
"rgbw_mode":"OFF",
"button_mode":"ON",
"button_pin":14,
"button_gy33":"OFF",
"ir_remote":"OFF",
"mqtt":"MQTT",
"home_assistant":"ON",
"legacy_animations":"ON",
"tv_animation":"ON",
"e131_animations":"ON",
"ota":"HTTP",
"state_save":"SPIFFS"}

Tell me how to understand this? I can not figure out the color
// parameters for automatically cycling favorite patterns
uint32_t autoParams[][6] = {   // main_color, back_color, xtra_color, speed, mode, duration (seconds)
  {0x00ff0000, 0x0000ff00, 0x00000000, 200,  1,  5000}, // blink red/geen for 5 seconds

bpohvoodoo, 2019-09-21:
0xwwrrggbb in HEX
for ultra white on WRGB
0xFFFFFFFF
for white on RGB
0x00FFFFFF
for full red 
0X00FF0000
for full green
0x0000FF00
for full blue
0x000000FF
or any other

viktor6, 2019-09-22:
How can I set timings correctly?
I want to make smooth transitions from one mode to another so that the modes are not interrupted in the middle. I have ws2812 for 180 lamps (3 meters x 60 led).

And in [GET] / status it does not show which mode is currently working in the car. To track the timing

```
uint32_t autoParams[][6] = {   // main_color, back_color, xtra_color, speed, mode, duration (seconds)
 // {0x00ff0000- red, 0x0000ff00- green, 0x000000ff- blue 
 //
  {0x00ff0000, 0x000000ff, 0x00000000, 50,  3, 10000},
  {0x0000ff00, 0x000000ff, 0x00000000, 50,  3, 10000},
  {0x00ff0000, 0x0000ff00, 0x00000000,  50, 4, 10000},
  {0x0000ff00, 0x000000ff, 0x00000000,  60, 6, 15000},
  {0x00ff0000, 0x0000ff00, 0x00000000,  60, 5, 15000},
  {0x00ff0000, 0x0000ff00, 0x000000ff,  80, 7, 15000},
  {0x00ff0000, 0x0000ff00, 0x000000ff,  100, 8,  15000},
  {0x00ff0000, 0x0000ff00, 0x000000ff,  100, 11, 15000},
  {0x00ff0000, 0x0000ff00, 0x000000ff,  100, 12, 15000},
  {0x00ff0000, 0x0000ff00, 0x000000ff,  100, 32, 15000},
  {0x00ff0000, 0x0000ff00, 0x000000ff,  100, 33, 15000},
  {0x00ff0000, 0x0000ff00, 0x000000ff,  100, 38, 15000},
  {0x00ff0000, 0x0000ff00, 0x000000ff,  100, 39, 15000},
  {0x00ff0000, 0x0000ff00, 0x000000ff,  50,  42, 15000}
};
```

bpohvoodoo, 2019-09-27:
If you want you can try the development branch now. Autoplay is saved in this version.

viktor6, 2019-09-27:
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

bpohvoodoo, 2019-09-28:
Hmm, that is weird, because I compiled on an NodeMCU, too.

viktor6, 2019-09-28:
Arduino IDE 1.8.10

bpohvoodoo, 2019-09-28:
I use  1.8.10, too.

viktor6, 2019-09-28:
reinstalled Arduino IDE 1.8.10 and esp8266 2. 5.2
The error remained

ACKET2009, 2019-09-28:
What adress do You paste for the esp resources? http, or https? I find the errors during compilation with the downloaded controller resources from https site - use the http.

viktor6, 2019-09-28:
Everything was fine, it was necessary to install the library //https://github.com/pasko-zh/brzo_i2c

viktor6, 2019-09-28:
And why not make the AUTOPLAY settings in the web config so that you can edit AUTOPLAY effects from the web page?

bpohvoodoo, 2019-09-28:
now autoplay is a normal animation
http://192.168.0.120/auto is obsolete now
it is http://192.168.0.120/set?m=56

bpohvoodoo, 2019-09-28:
> And why not make the AUTOPLAY settings in the web config so that you can edit AUTOPLAY effects from the web page?

So are may plans for future versions, but you are free to contribute code.

viktor6, 2019-09-28:
And now how to control speed and brightness?
http://192.168.0.120/set_brightness?p=1
File Not Found
URI: /set_brightness
Method: GET
Arguments: 1
 p: 1

bpohvoodoo, 2019-09-29:
Read the Rest-API docu V3
http://192.168.0.120/set?p=1
All set commands are replaced.

viktor6, 2019-09-30:
Check if it works for you in version V3.0.0.BETA1 auto?

ACKET2009, 2019-10-17:
Hello! The autoplay switch is excluded now from the web interface - only the iparess/set?m=56 work?

bpohvoodoo, 2019-10-20:
Autoplay is now a "normal" animation (56). You can select it via GUI like all the other animations.

ACKET2009, 2019-10-20:
Thank You! I use the development brunch. The version is V3.0.0.BETA3. After the first booting and making the wi-fi settings, the lights boot into static red definition. The auto switch on the web page is not present. Only after the string: ipadress/set?m=56 lights begin to autoswitching... If then the custom setting will be checked, the procedure with this string needed to repeat...

bpohvoodoo, 2019-10-20:
Beta Brach is 3.1.0.0BETA3

ACKET2009, 2019-10-20:
development and beta branch - the same files? I has downloaded from the development and write about that archive...

bpohvoodoo, 2019-10-20:
Yes beta and Development is the same (for the moment).

closed 2019-11-02 by bpohvoodoo