fablab-luenen/McLighting #32: Values for gamma correction and color picker not saved in SPIFFS Mode

Cypris2010, 2019-09-09:
**Describe the bug**
When MCLighting ist compiled with ENABLE_STATE_SAVE 1 some values (tested with gamma correction an color picker) are not saved when the device is shut down by power loss (removing power).

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the user interface
2. Click on the settings icon
3. Change the described values, press "save settings". Settings keep active as long the device is powered. After switching off an on power the settings are discarded.  
4. See error

SKETCH_VERSION "2.2.5.RU1.rgbw.3c"

Don't know if they are not saved ore not reloaded.

bpohvoodoo, 2019-09-10:
Maybe your SPIFFS is corrupt,
Try http://ipofyourdevice/format_spiffs
and check again.

bpohvoodoo, 2019-09-10:
Ok found a bug, causing settings were saved only for one time.
After reboot you can save again - for one time.
Missed to detach a ticker.
Will be corrected soon.

Cypris2010, 2019-09-11:
How can i debug the gamma value? It still seems not to reload this value?

Could it be possible that there is not enough space? I'm using a Wemos board.

bpohvoodoo, 2019-09-12:
Which gamma value do you mean?
Do you mean the value in the definitions, or do you mean the GAMMA setting in the GUI?
The gamma value in the definitions is for the a color sensor GY-33 and corrects its readings, if connected.
The GAMMA setting in the gui corrects the strips output to match the led colors on the WHEEL and CIRCLE input.

Cypris2010, 2019-09-12:
I mean the GAMMA setting in the GUI.

bpohvoodoo, 2019-09-13:
GAMMA is dooing as it should in my case. To check Please choose
Red value:255
Green: 55
Blue:0
White:0
Then turn on/off gamma. in gui. You should see a differnce.

bpohvoodoo, 2019-09-13:
Ok found the reason. I looked at saving which was fine, but the reading after reboot failed!

ryancasler, 2019-09-17:
OOOPS!  Doesn't do much good to save it if you're not using it later.  ;-)  thanks so much!!!

closed 2019-09-19 by bpohvoodoo