fablab-luenen/McLighting #39: IP address output

viktor6, 2019-09-24:
Draw the IP address that the wi-fi detected in Mqtt
This is necessary for the ripening of a smart home in a smart home

bpohvoodoo, 2019-09-25:
Can you please explain exactly what you mean?

viktor6, 2019-09-25:
home/McLighting1_ha/state/out
 {"state": "ON",**"ip": "192.168.2.123",**"color": {"r": 255,"g": 0,"b": 0,"w": 0,"r2": 0,"g2": 255,"b2": 0,"w2": 0,"r3": 0,"g3": 0,"b3": 255,"w3": 0},"brightness": 200,"color_temp": 327,"speed": 60,"effect": "AUTO"}

bpohvoodoo, 2019-09-26:
I do not see why you should need this.

viktor6, 2019-09-26:
It’s hard to explain, but believe me, this is useful for others.

viktor6, 2019-10-16:
IP definition does not work
{
  "state" : "ON",
  "ip" : true,
  "color" : {
    "r" : 255,
    "g" : 0,
    "b" : 0,
    "w" : 0,
    "r2" : 0,
    "g2" : 255,
    "b2" : 0,
    "w2" : 0,
    "r3" : 0,
    "g3" : 0,
    "b3" : 255,
    "w3" : 0
  },
  "brightness" : 126,
  "color_temp" : 327,
  "speed" : 122,
  "effect" : "Color Wipe"
}

bpohvoodoo, 2019-10-19:
I will have a look

viktor6, 2019-10-21:
Everything is working

closed 2019-10-21 by bpohvoodoo