fablab-luenen/McLighting #40: HHTP Rest API V3?

ryancasler, 2019-10-07:
Noticed that there was a new version of the HTTP Rest API in the wiki.  Does that mean that the commands have changed again?  I noticed that /all is no longer listed.  Did that change back to set_mode?

bpohvoodoo, 2019-10-07:
Yes the all command ist replace by /set?m=0 which was the same.
The /set_mode and all other set_* commands are replaced by set, because they didn't made sense.
No they are the command:
/config to manipulate all hole strip and esp specific settings.
/set to manipulate all segment specific settings.

ryancasler, 2019-10-07:
okay.  But you can still set multiple parameters in one request, correct?  So, currently I am using:

http://192.168.1.142/all?m=15&rgb=FF0000&rgb2=000000&s=180&c=100

I would just change "/all" to "/set" and leave the rest as is?

bpohvoodoo, 2019-10-08:
Yes.

closed 2019-10-08 by bpohvoodoo