fablab-luenen/McLighting #45: Some settings like led-count won't be saved..

tbone2k-git, 2019-10-12:
Hi there! o)

McLighting does not save led-count and the effect-state for me, maybe you have an idea why?

I compiled the nodemcu version and flashed it onto this pre-made SP501E controller:
https://www.aliexpress.com/item/33018042195.html

Initial wifi-setup in ap-mode worked fine, then I configured led-count to 120 and color to white 100% via the web-interface, then used the "save settings" button. 
After powering down and powering up the controller, the previous wifi connection and its setting is re-established fine, but the led-count is back to 60 and color is set to red 100%.

I'm using a GRBW SK6812 strip, powersupply is 5v@4amps, voltage is stable even if R+G+B+W is at 100%.

I'd appreciate any advice on how to tackle this problem.
Thank you in advance and have a nice day! o)

bpohvoodoo, 2019-10-20:
First of all: which version do you use?
Second there is a saving delay of about 3 seconds.
If you reset your device to early it will not save.

tbone2k-git, 2019-10-20:
Mhh.. I'm not exactly sure which version, I downloaded the zipped sources on October 9th and compiled these. I had a look in the web ui and in some source files, unfortunately I couldn't find a version string to tell you, but I hope "11 days old" is ok? o)

Yes, of course I waited some time before removing power and putting it back. Just tried again to make sure, but no. It'll always start with 50 leds and 100% red. (notice it's 50 leds (the default?) not 60 as in my first post).

Could it be the controller (or the 8285 esp on there) I bought does not support spiffs (this storage space thing)?. My knowlegde about the esp's is little right now, but I started reading and was looking through the code, somewhere I read eprom vs. spiffs usage. Does it make sense to look into this direction?

On the other hand, if ML is able to save wifi-settings successfully, why would it not save the led count and colors, so it's probably not related to the storage used, but this is just guessing of course. I had another quick test, the mqtt-server ip is not saved either.

Thank you! o)

bpohvoodoo, 2019-10-20:
Please use http://yourdeviceip/esp_status then you will get a version string and more.
If you have no spiffs, saving will not work! after version 2.2.7.
In version 2.2.7 you have to change
#define ENABLE_STATE_SAVE 1
to
#define ENABLE_STATE_SAVE 0

bpohvoodoo, 2019-10-20:
Try this setting
[image]

tbone2k-git, 2019-10-21:
Hi bpohvoodoo, thank you for your support! o)
Your advice helped me to make some progress! o)

I cloned the repo once again and compiled/uploaded via platformio this time (I just used esptool.py before). Now I'm using 2.2.7, I did not change any definitions and changing the led count seems to be persistent now. Power-cycling the controller does not bring up defaults anymore. So this is good! o)

I'm still confused, because then I set the correct button-pin in the definition, to be able to use the physical button on the controller. When I press the button, the led-strip will turn off, pressing it again will not bring up the last state/color of the strip, it will always be white and "static" mode. Is this expected behaviour? I would expect the button to work the same as power-cycling the controller. 

Thanks again!

ps:
I did not use ArduinoIDE yet, I'm just making my first steps with all this and was happy that I got PlatformIO working recently. Although I don't seem to like the VSCode+PlatformIO bloat at all. If I experience further issues or you really suggest sticking to ArduinoIDE I will go for that! o)

tbone2k-git, 2019-10-21:
It's me again! o)

I had a look through the code to see what happens when the button is pressed. All is clear now, button presses are "hard wired" to pre-defined states. So it seems everything is working as designed, which is not necessarily as expected by me, but no problem! o)

I have learned something, thank you! o)

closed 2019-10-21 by tbone2k-git