fablab-luenen/McLighting #57: After the web tuning single mode autoplay mode disabled

ACKET2009, 2019-11-12:
Hello! The version, I use, is latest - beta 5
The autoplay mode starts normally after flashing - it's ok!
While I make any definition on the web interface - m.b. static, after the restart it is always in static mode. And without this string http://192.168.1.6/set?m=56 on the web browser no any choice to start the autoplaying again. There is no switch to toggle off/on autoplay on the web page now. Please return it!

ACKET2009, 2019-11-13:
Or, may be, make the autoplay mode by default, after each reconnect or reset...

bpohvoodoo, 2019-11-15:
Try using this (Animation 56):
[image]

closed 2019-11-17 by bpohvoodoo