fablab-luenen/McLighting #61: All Light Params in One Request to HTTP API

ryancasler, 2019-12-13:
It's possible that this is already working but I could not find a way to retrieve all of the current settings in one HTTP request.  Using the endpoint /all turns them back on if they were previously switched off, so that doesn't really work all that well.  I really don't want to have to go through the trouble of issuing 5 or 6 http calls to get all of the parameters from the board.  Thanks!

bpohvoodoo, 2019-12-14:
Maybe you want to give /status a try

https://github.com/FabLab-Luenen/McLighting/wiki/HTTP-REST-API-(V3)

ryancasler, 2019-12-15:
DUH!   We have an expression in the US, "if it was a dog, it would have bit me."  I looked at that page for almost an hour...not of course it stands right out.  Thanks for the tip!

closed 2019-12-15 by ryancasler