fablab-luenen/McLighting #67: Don't publish status change on MQTT-out when change was from MQTT-in

mikkel75, 2020-01-29:
Hi,

Great project btw!

I have an issue when I want to be able to monitor and control McLighting devices from my Domoticz system using MQTT. My system looks like this
https://1drv.ms/u/s!AlY-U_GE-GiVgsJYYFZF4eo25_vGwg?e=Ivfnb6

The problem is that I get a MQTT message loop, since both McLighting and Domoticz sends out a new MQTT message when it receives one.

I have discussed with the Domoticz developer about a way to update a virtual device WITHOUT triggering a new MQTT message. So far with no luck.

It would be great if McLighting could have such a feature, so that state change based on incomming MQTT wouldn't result in a MQTT out message.

Hope you can help :)

Thanks

bpohvoodoo, 2020-02-02:
Hi I will have a look. Possibly i will add a compile sitch for this into the definitions.h file

mikkel75, 2020-02-04:
Hi.
Sounds good....just out of curiosity - is it normal for MQTT devices to transmit it's status after a change from MQTT/in ? If so, I would agree that the problem lies with Domoticz...

bpohvoodoo, 2020-02-04:
Yes, it is. Otherwise you can't be shure the device did what you wanted.

closed 2020-03-07 by bpohvoodoo