fablab-luenen/McLighting #70: MQTT seems to be unstable.

Allday3D, 2020-02-22:
**Describe the bug**
A clear and concise description of what the bug is, is probably not possible because there seems to be no logic in it. MQTT on the device can work for days very well and the other day it's just don't.
The only feedback (on the config topic) seems to be an OFFLINE message. Only a power off/on can bring the MQTT back.
I know for sure that the broker/server on the Raspberry Pi is working because all other devices work perfectly and are also visible in MQTT.fx on a Windows machine.

**To Reproduce**
Steps to reproduce the behavior:
1. I have #define ENABLE_OTA 0 and #define ENABLE_MQTT 0 uncomment but no MQTT_HOSTNAME, no HOMEASSISTANT and no MQTT_HOME_ASSISTANT_SUPPORT.
2. Compiled on a NodeMCU
3. Runs for a couple of days.
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen is that the MQTT just keeps working or at least try's to reconnect by itself.

**Additional context**
Add any other context about the problem I don't have.
only like to point to the fact that the MQTT pasword is visible in the console of Firefox.

bpohvoodoo, 2020-02-22:
What version do you use, because there may be a heap problem in V2.
Please try Beta branch

Allday3D, 2020-02-22:
Seems I use the Master (the default one). I will try the Beta.

Allday3D, 2020-03-10:
After running the beta version now for more then 2 weeks I think the MQTT is much more stable. The only problem I seem to have now is that when using the button function it only sends the speed  (? 255) information over MQTT.

bpohvoodoo, 2020-03-10:
Maybe you should use the homeassistant topics, even if you are not using homeassistant, 
but i will have a look.

ryancasler, 2020-07-14:
I am having a similar problem with MQTT.  I am using pub/sub not the async library and it will randomly just stop reporting back changes on the out channel.  I have had to implement a nightly reboot of the board in order to resolve this.

bpohvoodoo, 2020-07-19:
Will check this.

ryancasler, 2020-09-02:
Any info on this?  It continues to get worse and worse.  I am rebooting every 4 hours now to keep connectivity alive.  If sending the "off" payload didn't turn the strip back on if it was already off (a different issue listing for that one is already listed) then I could try re-sending the "off" to keep connectivity.  But as it stands there is nothing to be done.

bpohvoodoo, 2020-09-05:
Haven't had any time yet.

ryancasler, 2020-11-21:
I don't know if any of the changes implemented in the latest release had anything to address this specifically but after 24 hours, it seems more stable now.  No disconnects yet.  Will let you know if I finally see one.

bpohvoodoo, 2020-11-21:
By now I know why it crashes after a time, but I don't know how to fix at the moment. The heap is the problem, again. It runs out of memory.

ryancasler, 2020-11-29:
Well, I don't know what to tell you...something has gotten better.  It's been a week since I rebooted and no MQTT disconnects that I can detect.  Maybe the new version of the ESP8266 wifi library is playing nicer with the PubSubClient library.  Or maybe it is resetting itself when it happens now. 🤷‍♂️  All I know is that before I couldn't go 24 hours without having to manually reboot the board and now it's been 7 days.  I felt I should let you know since I was one of the ones complaining about it before.  It only seems fair if someone is going to complain that they spend at least as much time pointing out when things go well.  Right?  👍

bpohvoodoo, 2020-11-30:
I will keep an eye on it.