fablab-luenen/McLighting #75: Led's not lighting up

andrein81, 2020-04-06:
I install it more devices, try with different LED strips but not working

The web UI is working.

bpohvoodoo, 2020-04-07:
More information would be useful.
Please post http://{ip_of_device}/esp_status

andrein81, 2020-04-11:
{"HOSTNAME":"McLightingRGBW","version":"3.1.0","heap":19248,"sketch_size":626416,"free_sketch_space":1470464,"flash_chip_size":4194304,"flash_chip_real_size":4194304,"flash_chip_speed":40000000,"sdk_version":"2.2.2-dev(38a443e)","core_version":"2_6_3","cpu_freq":80,"chip_id":1458208,"animation_lib":"WS2812FX_DMA","ws2812_pin":3,"led_count":50,"rgb_order":"GRBW","rgbw_mode":"ON","button_mode":"OFF","button_gy33":"OFF","ir_remote":"ON","tsop_ir_pin":13,"mqtt":"AMQTT","home_assistant":"ON","ota":"HTTP","state_save":"SPIFFS"}

bpohvoodoo, 2020-04-12:
Ok, so you use an RGBW Strip (maybe SK6812) connected to GPIO03 which is the RX Pin, right?

andrein81, 2020-04-12:
no, I have a ws2812 on D4. I change in the definitions.h the #define LED_PIN 2. Even on the web UI it shows pin 3 and it cannot be change. Not sure what I do wrong :(

andrein81, 2020-04-12:
#define USE_WS2812FX_DMA 2      // 0 = Used PIN is ignored & set to RX/GPIO3; 1 = Used PIN is ignored & set to TX/GPIO1; 2 = Used PIN is ignored & set to D4/GPIO2;  Uses WS2812FX, see: https://github.com/kitesurfer1404/WS2812FX
                                // or comment it out
#if defined(USE_WS2812FX_DMA)
  #define MAXLEDS 384           // due to memory limit of esp8266 at the moment only 384 leds are supported in DMA Mode. More can crash if mqtt is used.
#else
  #define MAXLEDS 4096
#endif
// Neopixel
#define LED_PIN 2          // PIN (15 / D8) where neopixel / WS2811 strip is attached; is configurable, if USE_WS2812FX_DMA is not defined. Just for the start
#define NUMLEDS 50         // Number of leds in the; is configurable just for the start
#define RGBORDER "GRBW"    // RGBOrder; is configurable just for the start
#define FX_OPTIONS 48      // ws2812fx Options 48 = SIZE_SMALL + FADE_MEDIUM  is configurable just for the start; for WS2812FX setSegment OPTIONS, see: https://github.com/kitesurfer1404/WS2812FX/blob/master/extras/WS2812FX%20Users%20Guide.md
#define LED_TYPE_WS2811    // Uncomment, if LED type uses 400 KHz (classic 'v1' (not v2) FLORA pixels, WS2811 drivers)
#define LED_BUILTIN 2      // ESP-12F has the built in LED on GPIO2, see https://github.com/esp8266/Arduino/issues/2192
char HOSTNAME[65] = "NedelcuLEDtest"; // Friedly hostname  is configurable just for the start. Hostname should not contain spaces as this can break Home Assistant discovery if used.

#define ENABLE_OTA 1                  // If defined, enable Arduino OTA code. If set to 0 enable Arduino OTA code, if set to 1 enable ESP8266HTTPUpdateServer OTA code.
#define ENABLE_MQTT 1                 // If defined use MQTT OR AMQTT, if set to 0 enable MQTT client code, see: https://github.com/toblum/McLighting/wiki/MQTT-API, if set to 1, enable Async MQTT code, see: https://github.com/marvinroger/async-mqtt-client
//#define ENABLE_MQTT_HOSTNAME_CHIPID   // Uncomment/comment to add ESPChipID to end of MQTT hostname
//#define ENABLE_MQTT_INCLUDE_IP        // uncomment/comment to add the IP-adress to the MQTT message
#define ENABLE_HOMEASSISTANT          // If defined, enable Homeassistant integration, ENABLE_MQTT must be active
#define MQTT_HOME_ASSISTANT_SUPPORT   // If defined, use AMQTT and select Tools -> IwIP Variant -> Higher Bandwidth
//#define DISABLE_MQTT_OUT_ON_MQTT_IN   // If defined, McLighting will not send back MQTT-out on MQTT-in regarding issue #67, does not change anything at the moment

//#define ENABLE_BUTTON 14              // If defined, enable button handling code, see: https://github.com/toblum/McLighting/wiki/Button-control, the value defines the input pin (14 / D5) for switching the LED strip on / off, connect this PIN to ground to trigger button.
//#define ENABLE_BUTTON_GY33 12         // If defined, enable button handling code for GY-33 color sensor to scan color. The value defines the input pin (12 / D6) for read color data with RGB sensor, connect this PIN to ground to trigger button.
//#define POWER_SUPPLY 12               // PIN (12 / D6) If defined, enable output to control external power supply
#if defined(POWER_SUPPLY)
  #define POWER_ON   HIGH           // Define the output state to turn on the power supply, either HIGH or LOW.  Opposite will be uses for power off.
#endif
#define ENABLE_REMOTE 13              // If defined, enable Remote Control via TSOP31238. The value defines the input pin (13 / D7) for TSOP31238 Out

#if defined(ENABLE_BUTTON_GY33)
  #define GAMMA 2.5                   // Gamma correction for GY-33 sensor
#endif

#define ENABLE_STATE_SAVE             // If defined, load saved state on reboot and save state on SPIFFS

#define CUSTOM_WS2812FX_ANIMATIONS    // uncomment and put animations in "custom_ws2812fx_animations.h"
#define USE_HTML_MIN_GZ               // uncomment for using index.htm & edit.htm from PROGMEM instead of SPIFFS

#define TRANS_COLOR_DELAY 5            // Delay for color transition
#define TRANS_DELAY 10                 // Delay for brightness and speed transition

bpohvoodoo, 2020-04-12:
Then please try either disabling DMA by commenting out
//#define USE_WS2812FX_DMA 2
the you can select any GPIO Pin via UI
or try 
#define USE_WS2812FX_DMA 0
and connect you strip to RX/GPIO03.
In the UI you hae to select GRB color order not GRBW

andrein81, 2020-04-13:
Thank you very much. I manage to make it work. Below is the definitions.h that worked for me. I use pin D6

//#define USE_WS2812FX_DMA 0      // 0 = Used PIN is ignored & set to RX/GPIO3; 1 = Used PIN is ignored & set to TX/GPIO1; 2 = Used PIN is ignored & set to D4/GPIO2;  Uses WS2812FX, see: https://github.com/kitesurfer1404/WS2812FX
                                // or comment it out
#if defined(USE_WS2812FX_DMA)
  #define MAXLEDS 384           // due to memory limit of esp8266 at the moment only 384 leds are supported in DMA Mode. More can crash if mqtt is used.
#else
  #define MAXLEDS 4096
#endif
// Neopixel
#define LED_PIN 12          // PIN (15 / D8) where neopixel / WS2811 strip is attached; is configurable, if USE_WS2812FX_DMA is not defined. Just for the start
#define NUMLEDS 50         // Number of leds in the; is configurable just for the start
#define RGBORDER "GRB"    // RGBOrder; is configurable just for the start
#define FX_OPTIONS 48      // ws2812fx Options 48 = SIZE_SMALL + FADE_MEDIUM  is configurable just for the start; for WS2812FX setSegment OPTIONS, see: https://github.com/kitesurfer1404/WS2812FX/blob/master/extras/WS2812FX%20Users%20Guide.md
#define LED_TYPE_WS2811    // Uncomment, if LED type uses 400 KHz (classic 'v1' (not v2) FLORA pixels, WS2811 drivers)
#define LED_BUILTIN 2      // ESP-12F has the built in LED on GPIO2, see https://github.com/esp8266/Arduino/issues/2192
char HOSTNAME[65] = "NedelcuLEDtest"; // Friedly hostname  is configurable just for the start. Hostname should not contain spaces as this can break Home Assistant discovery if used.

#define ENABLE_OTA 1                  // If defined, enable Arduino OTA code. If set to 0 enable Arduino OTA code, if set to 1 enable ESP8266HTTPUpdateServer OTA code.
#define ENABLE_MQTT 1                 // If defined use MQTT OR AMQTT, if set to 0 enable MQTT client code, see: https://github.com/toblum/McLighting/wiki/MQTT-API, if set to 1, enable Async MQTT code, see: https://github.com/marvinroger/async-mqtt-client
//#define ENABLE_MQTT_HOSTNAME_CHIPID   // Uncomment/comment to add ESPChipID to end of MQTT hostname
//#define ENABLE_MQTT_INCLUDE_IP        // uncomment/comment to add the IP-adress to the MQTT message
#define ENABLE_HOMEASSISTANT          // If defined, enable Homeassistant integration, ENABLE_MQTT must be active
#define MQTT_HOME_ASSISTANT_SUPPORT   // If defined, use AMQTT and select Tools -> IwIP Variant -> Higher Bandwidth
//#define DISABLE_MQTT_OUT_ON_MQTT_IN   // If defined, McLighting will not send back MQTT-out on MQTT-in regarding issue #67, does not change anything at the moment

//#define ENABLE_BUTTON 14              // If defined, enable button handling code, see: https://github.com/toblum/McLighting/wiki/Button-control, the value defines the input pin (14 / D5) for switching the LED strip on / off, connect this PIN to ground to trigger button.
//#define ENABLE_BUTTON_GY33 12         // If defined, enable button handling code for GY-33 color sensor to scan color. The value defines the input pin (12 / D6) for read color data with RGB sensor, connect this PIN to ground to trigger button.
//#define POWER_SUPPLY 12               // PIN (12 / D6) If defined, enable output to control external power supply
#if defined(POWER_SUPPLY)
  #define POWER_ON   HIGH           // Define the output state to turn on the power supply, either HIGH or LOW.  Opposite will be uses for power off.
#endif
#define ENABLE_REMOTE 13              // If defined, enable Remote Control via TSOP31238. The value defines the input pin (13 / D7) for TSOP31238 Out

#if defined(ENABLE_BUTTON_GY33)
  #define GAMMA 2.5                   // Gamma correction for GY-33 sensor
#endif

#define ENABLE_STATE_SAVE             // If defined, load saved state on reboot and save state on SPIFFS

#define CUSTOM_WS2812FX_ANIMATIONS    // uncomment and put animations in "custom_ws2812fx_animations.h"
#define USE_HTML_MIN_GZ               // uncomment for using index.htm & edit.htm from PROGMEM instead of SPIFFS

#define TRANS_COLOR_DELAY 5            // Delay for color transition
#define TRANS_DELAY 10                 // Delay for brightness and speed transition

closed 2020-04-13 by bpohvoodoo