fablab-luenen/McLighting #77: NodeMcu not start

manuallv, 2020-04-13:
All settings by default. I don't know how to even name this error?
Can anyone help?

1. upload all content. ok. but nothing happens next
[image]

2. When restart Nodemcu module then, serial Monitor show me this:
[https://youtu.be/2Ku9qin4zII](url)

where could be the problem?

bpohvoodoo, 2020-04-13:
It seems, that you already used this NodeMCU for Toblums original Project.
Please erase all content with:
[image]

manuallv, 2020-04-13:
> It seems, that you already used this NodeMCU for Toblums original Project.
> Please erase all content with:
> [image]

Thanks, everything is working now

closed 2020-04-14 by bpohvoodoo