fablab-luenen/McLighting #80: MQTT Out Message Not Accurate When Returning to Previously Selected Mode

ryancasler, 2020-04-15:
When monitoring the MQTT /out topic, if the controller is set to go back to the previously selected mode, the payload sent is still "OK =off".  Only when the newly selected mode is something other than what was previously selected does the topic change correctly.

To reproduce this problem:
1. Go to the web console.
2. Select any combination of Colors/brightness and mode= Static
3. Select Off (this will produce the payload "OK =off" as expected)
4. Select mode Static again.  (This will also produce the payload "OK =off".  I would expect this to return "OK /0".)

This works for any combination of mode/color and brightness.  Simply selecting off and then returning to the previously selected mode produces the wrong payload.

ryancasler, 2020-04-15:
It also appears that I am unable to get regular MQTT messages when using ENABLE_HOMEASSISTANT.  The only way I can get status updates to be sent to my broker is to use the Aysync method but that results in a rather significant delay between the change taking place and the message being sent to my broker.  

I am trying to use the message to detect the state of the controller since I have two possible control points outside of my McLighting controller.  Rather than go through the trouble of colating them before being sent to the controller I figured I could just use the MQTT messages to detect a change initiated by the other system.  But when I subscribe to home/<HOSTNAME>_ha/state/out, I only get messages when using the ASYNC method.  Any assistance would be greatly appreciated.  

This is the relevant section of my definitions.h file.  

#define ENABLE_OTA 1                  
#define ENABLE_MQTT 0               
//#define ENABLE_MQTT_HOSTNAME_CHIPID   
//#define ENABLE_MQTT_INCLUDE_IP        
#define ENABLE_HOMEASSISTANT          
//#define MQTT_HOME_ASSISTANT_SUPPORT  
//#define DISABLE_MQTT_OUT_ON_MQTT_IN   

If you could let me know how to fix either of these issues, that would be a huge help.

bpohvoodoo, 2020-04-15:
I will have a look.

bpohvoodoo, 2020-04-15:
First i can not reproduce.

> To reproduce this problem:
>
> Go to the web console.
> Select any combination of Colors/brightness and mode= Static
> Select Off (this will produce the payload "OK =off" as expected)
> Select mode Static again. (This will also produce the payload "OK =off". I would expect this to return "OK /0".)
> This works for any combination of mode/color and brightness. Simply selecting off and then returning to the previously selected mode produces the wrong payload.

Would you please post http://{ip_of_your_device}/esp_status.

The second not updating homeassistant using PubSubClient i can reproduce.

FYI: There is a 2-3 sec delay if you use AsyncMqtt. This can be reduced to almost 0, but is not recommended.

ryancasler, 2020-04-15:
Sure.  Here's the esp_status:

`{"HOSTNAME":"deskMclighting","version":"3.1.0","heap":21160,"sketch_size":558656,"free_sketch_space":2584576,"flash_chip_size":4194304,"flash_chip_real_size":4194304,"flash_chip_speed":40000000,"sdk_version":"2.2.1(cfd48f3)","core_version":"2_4_2","cpu_freq":80,"chip_id":1458415,"animation_lib":"WS2812FX_DMA","ws2812_pin":3,"led_count":50,"rgb_order":"GRB","rgbw_mode":"OFF","button_mode":"ON","button_pin":14,"button_gy33":"OFF","ir_remote":"OFF","mqtt":"AMQTT","home_assistant":"ON","ota":"HTTP","state_save":"SPIFFS"}`

> There is a 2-3 sec delay if you use AsyncMqtt.

I'm seeing more of a 5-7 second delay.  Could that is be because of my setup?  Do I need to change one of the settings for QoS (i'm still wrapping my head around the MQTT).

> The second not updating homeassistant using PubSubClient i can reproduce.

Oh good...so I'm not going crazy(er)  😉 

> First i can not reproduce.

Here are the logs from Node Red.  You can clearly see, first both the /out and /ha/state/out both report off.  Then I turn it to static mode by clicking on static in the WEB Gui.  (That and the HTTP API are the only two control methods I have tried and confirmed this issue with).  You can see /out goes to "OK /0" and the same is reported to HA.  Then I turn it off by clicking off in the web GUI.  Again, both statuses show off correctly.
Next, I click "Static" again in the web GUI.  You can clearly see the /out reports "OK =OFF" and HA reports "State": ON.  This is where the problem lies.

[image]

Should the last report to /out really be "OK =off"?  The strip is on at that point.

bpohvoodoo, 2020-04-18:
Fixed HA not working with PubSubClient

bpohvoodoo, 2020-04-18:
Ahh, now i can reproduce, but i have no solution yet.
Stand by.....

ryancasler, 2020-04-18:
> Ahh, now i can reproduce, but i have no solution yet.

I was about to get out my phone and record it for you.   I might be going Covid-crazy but not THAT crazy.  ;-)  Thanks for all your help!

ryancasler, 2020-04-19:
> Fixed HA not working with PubSubClient

I'm still not able to get HA MQTT messages if I only enable:
#define ENABLE_HOMEASSISTANT 
Maybe I am not understanding the intended functionality.  But I thought that enabling this would get HA communications using MQTT.  And that enabling:
#define MQTT_HOME_ASSISTANT_SUPPORT
would get auto-discovery of HA devices.  But either that is incorrect or it is not working as intended.  Because I get nothing out unless I enable the second option as well.  In fact, if I have my settings like this:

#define ENABLE_MQTT 0                 
//#define ENABLE_HOMEASSISTANT         
//#define MQTT_HOME_ASSISTANT_SUPPORT  
#define DELAY_MQTT_HA_MESSAGE 2  

I get MQTT messages on <HOSTNAME>/out.  They're wrong, but I get them.  Then if I change the settings to this:

#define ENABLE_MQTT 0                 
#define ENABLE_HOMEASSISTANT         
//#define MQTT_HOME_ASSISTANT_SUPPORT  
#define DELAY_MQTT_HA_MESSAGE 2  

I then get no MQTT messages at all, either on <HOSTNAME>/out or on home/<HOSTNAME>_ha/state/out.  It is only when I change the settings to:

#define ENABLE_MQTT 0                 
#define ENABLE_HOMEASSISTANT         
#define MQTT_HOME_ASSISTANT_SUPPORT  
#define DELAY_MQTT_HA_MESSAGE 2  

do I start to get MQTT messages again.  So, am I totally misunderstanding how those setting are supposed to work or is something seriously wrong here?  Because, I thought you had to have AMQTT enabled for MQTT_HOME_ASSISTANT_SUPPORT  but it is working with PUBSUB.
(note: I removed the extra settings that did not change but the other settings are at default)

I have everything I need now that you have the customizable delay option available.   This is purely for my edification that it just doesn't seem right.  But "That's how it works."  would be a completely acceptable answer at this point.  Don't want you to think I'm harping or nagging.  It just didn't look right to me and I know that kinda stuff bugs the hell out of me (pun totally intended) so I thought I would bring it up.  😄

bpohvoodoo, 2020-04-19:
@ryancasler 
Not finished yet but please test the development tree, if the behaviour is now as expected!!

ryancasler, 2020-04-20:
YAY!!!  You rock. I checked all the permutations I could think of and everything seems to be working per the docs now. Great work! Thank you!!!

closed 2020-04-20 by ryancasler