fablab-luenen/McLighting #86: 'main_color.white' always overwritten by 'white_value'

ChristophA, 2020-05-03:
**Describe the bug**
I'm using the current Development branch (3.1.2). I've anabled the HA integration.

I tried to light the attached RGBW strip in warm white. When sending a value for "w" to 'home/hostname_ha/state/in' then strip is not lighing up. Values for 'w2' and 'w3' are correctly set, but the value for 'w' is still 0. As far as I could analyse the value set for 'white_value' is overriding any value set for 'w'.
```
{
    "brightness": 125,
    "color": {
        "r": 0,
        "g": 0,
        "b": 0,
        "w": 255,
        "r2": 0,
        "g2": 0,
        "b2": 0,
        "w2": 255,
        "r3": 0,
        "g3": 0,
        "b3": 0,
        "w3": 255
    },
    "white_value": 0,
    "speed": 50,
    "effect": "Static",
    "state": "ON"
}
```
**Expected behavior**
Don't now if the behavior is as designed and intended, but I would prefer to not overwrite values for 'w' with values set for 'white_value'. On the other hand I could live with the current behavior, but it would be good to highlight this in the wiki section for the HA integration accordingly.

By the way, if sending values for 'color_temp' to 'home/hostname_ha/state/in' any value for 'r', 'g' and 'b' will be overwritten. That should be highlighted in the wiki, too.

Last but not least: Many thanky for all your efforts! Much appreciated! Great project!

bpohvoodoo, 2020-10-25:
I had a look in the code.
w and white_value are basically the same.
If you don't want white_value to overwrite w value, then don't send white_value.
As white_value is handled after w value it will always overwrite w value, if it is sent.

ChristophA, 2020-10-25:
Thanks for your feedback. As I now know how to handle this the issue can be closed.

closed 2020-10-25 by ChristophA