fablab-luenen/McLighting #89: No AP after flash to NodeMCU V3 ( Lolin ) because of Exception 28 Reset

zkidez, 2020-06-15:
**Describe the bug**
Ive used the VSCode methode to flash the NodeMCU, some little compiling issues were fixed by add Arduino.h lib .

Now after successfully flash the nodeMCU there is no AP, the debug Serial is telling me an exception  with code (28) after the "*WM: freeing allocated params!" step

Did you have ever seen such a case?
it should also be enough memory space there 
RAM:   [=====     ]  52.7% (used 43176 bytes from 81920 bytes)
Flash: [======    ]  59.6% (used 622204 bytes from 1044464 bytes)

> Starting...Main Setup
> FS File: /config.json, size: 150B
> FS File: /stripstate.json, size: 39B
> FS File: /stripstate_segment_00.json, size: 90B
> FS Usage: 1506/1953282 bytes
> Reading config file... Opened!
> Config: Parsed!
> {"hostname":"Lumos","mqtt_host":"","mqtt_port":1883,"mqtt_user":"","mqtt_pass":"","ws_seg":1,"ws_cnt":100,"ws_rgbo":"GRB","ws_pin":2,"ws_trans":false}
> Checking RGB Order: GRB ...success!
> WiFiManager config FS read success!
> Reading state file... Opened!
> Config: Parsed{"segment":0,"mode":2,"brightness":196}
> Strip state config FS read Success!
> *WM: Adding parameter
> *WM: hostname
> *WM: Adding parameter
> *WM: host
> *WM: Adding parameter
> *WM: port
> *WM: Adding parameter
> *WM: user
> *WM: Adding parameter
> *WM: pass
> *WM: Adding parameter
> *WM: strip_size
> *WM: Adding parameter
> *WM: rgbOrder
> *WM: Adding parameter
> *WM: fxoptions
> *WM:
> *WM: AutoConnect
> *WM: Connecting as wifi client...
> *WM: Status:
> *WM: 3
> *WM: Using last saved values, should be faster
> *WM: Connection result: 
> *WM: 3
> *WM: IP Address:
> *WM: 192.168.2.137
> Checking RGB Order: GRB ...success!
> connected...yeey :)
> Open http://192.168.2.137/ to open McLighting.
> 
> Initializing strip!
> Checking RGB Order: GRB ...success!
> Reading segmentstate file: /stripstate_segment_00.json
> Opened!
> Config: Parsed{"start":0,"stop":99,"fx_mode":0,"speed":0,"color":[0,0,0,0,0,0,0,0,0,0,0,0],"ws_fxopt":0}
> Segment state config FS read Success!
> Number of Segments: 1
> Listening for data...
> finished Main Setup!
> *WM: freeing allocated params!
> 
> Exception (28):
> epc1=0x40216805 epc2=0x00000000 epc3=0x00000000 excvaddr=0x000000d8 depc=0x00000000
> 
> >>>stack>>>
> 
> ctx: cont
> sp: 3fff2540 end: 3fff2790 offset: 01a0
> 3fff26e0:  4022cd69 00000001 3fff4cac 4021084c  
> 3fff26f0:  3fff4eac 00000001 3fff4cac 40210911
> 3fff2700:  ffffff80 00000000 00001388 000000c4
> 3fff2710:  00000000 00000000 3fff0630 3fff0f58
> 3fff2720:  3ffe8588 3ffe8578 3fff05fc 4020ffbe
> 3fff2730:  00000000 00000000 33383831 3031ef00
> 3fff2740:  feef0030 00425247 003834fe feefeffe
> 3fff2750:  00000001 feefeffe feefeffe feefeffe  
> 3fff2760:  feefeffe feefeffe feefeffe 3fff11a0
> 3fff2770:  3fffdad0 00000000 3fff1160 4022c684
> 3fff2780:  feefeffe feefeffe 3ffe88dc 40100589
> <<<stack<<<
> 
>  ets Jan  8 2013,rst cause:2, boot mode:(3,7)
> 
> load 0x4010f000, len 1392, room 16 
> tail 0
> chksum 0xd0
> csum 0xd0
> v3d128e5c
> ~ld

**To Reproduce / changes in definitions.h**

> #include <Arduino.h>    // <- stdin / unint8_t compiling error fix
> void initMqtt(void);     // <- Out of scope compiling error fix
> 
> #define USE_WS2812FX_DMA 2      // 0 = Used PIN is ignored & set to RX/GPIO3; 1 = Used PIN is ignored & set to TX/GPIO1; 2 = Used PIN is ignored & set to D4/GPIO2;  Uses WS2812FX, see: https://github.com/kitesurfer1404/WS2812FX
>                                
> #if defined(USE_WS2812FX_DMA)
>   #define MAXLEDS 384           
> #else
>   #define MAXLEDS 4096
> #endif
> // Neopixel
> #define LED_PIN 2        
> #define NUMLEDS 100       
> #define RGBORDER "GRB"   
> #define FX_OPTIONS 48     
> //#define LED_TYPE_WS2811   
> #define LED_BUILTIN 12  

> char HOSTNAME[65] = "Lumos";
> 
> #define ENABLE_OTA 1                 
> #define ENABLE_MQTT 1                
> //#define ENABLE_MQTT_HOSTNAME_CHIPID   
> //#define ENABLE_MQTT_INCLUDE_IP        
> #define ENABLE_HOMEASSISTANT          
> #define MQTT_HOME_ASSISTANT_SUPPORT  
> #define DELAY_MQTT_HA_MESSAGE 5

bpohvoodoo, 2020-06-15:
Have you Updated from V2?
If so Erase all flash content:
[image]

zkidez, 2020-06-15:
You are right ! 
This topic with the WiFi Settings in another memory area has helped me , i was using the board before with PubSubclient 

Many thanks for your quick help!

is it possible to set this in VSCode as default?

closed 2020-06-15 by zkidez