fablab-luenen/McLighting #90: Cant compile, help pls

huge-11, 2020-06-15:
Hello,
i want to try out this fork, but i cant compile. I am probably missing some libs, can anyone help me out? I added the libs in the wiki, but somehow its not working.
i get this error,

C:\Users\Documents\ESP8266\Glowballs\McLighthingV3\McLighting-master\McLighting-master\Arduino\McLighting\McLighting.ino: In function 'void loop()':

McLighting:726: error: 'handleE131Play' was not declared in this scope

     if (playE131 == true) { handleE131Play(); }

                                            ^

exit status 1
'handleE131Play' was not declared in this scope

thanks

bpohvoodoo, 2020-06-15:
This seems not to be a library problem.
The function is declared in "mode_custom_ws2812fx_animations.h"
Which Arduino Version do you use?

zkidez, 2020-06-15:
"'handleE131Play' was not declared in this scope"

i had the same issue just before with " initMQTT();" and not in scope

i just added          void  initMQTT(void);        to definition.h and it was working ( just as a "walk " around )

Just check what     handleE131Play (); has as return value ( uint8_t or something else ) and check what is input ( i think its void ) 
and write the so called called fuctionprototype into .h file

( ive just looked into the files, in this project as from VSCode loaded  there is no handleE131Play function, so maybe you have a newer version of E131 Lib)

huge-11, 2020-06-15:
My bad, i commented out the CUSTOM_WS2812FX_ANIMATIONS part in definitions.h, but thanks for the help.

closed 2020-06-15 by huge-11