fablab-luenen/McLighting #102: >450 LED (500 needed)

garfield2412, 2020-10-22:
I got a little issue,
It seems you can only adress 450 LED. If you config more, the +450 LEDS aren't lighting.
 I've read another guy hat the issue with >466LED and he tweaked the code so he got 600.
But he didn't clearly told how he tweaked it.
Someone wrote to uncomment "DMA" is a nice tweak for this but it didnt help.

Also tried maxled=500, but it didnt work for me - 450 is max.

I've installed 500 LED and it looks great, but i don't like it's stopps 1m before the rnd of the Wall :-(

I tried to start at LED 50, but it doesnt bump up the other LED's

bpohvoodoo, 2020-10-24:
Please send your
http://your_device_ip/esp_status

garfield2412, 2020-10-28:
> http://your_device_ip/esp_status
{"HOSTNAME":"McLightingRGBW","version":"3.1.1","heap":23632,"sketch_size":565072,"free_sketch_space":1531904,"flash_chip_size":4194304,"flash_chip_real_size":4194304,"flash_chip_speed":40000000,"sdk_version":"2.2.1(cfd48f3)","core_version":"2_4_2","cpu_freq":160,"chip_id":1451458,"animation_lib":"WS2812FX","ws2812_pin":3,"led_count":450,"rgb_order":"GRB","rgbw_mode":"OFF","button_mode":"OFF","button_gy33":"OFF","ir_remote":"ON","tsop_ir_pin":13,"mqtt":"OFF","home_assistant":"OFF","ota":"HTTP","state_save":"SPIFFS"}

bpohvoodoo, 2020-10-30:
At the moment i have no clue what this can be. And you are shure your strip is ok?
I have no strip of over 450 LED here for testing. Heap is ok so it is not a memory problem.
Maybe you try the 3.1.3 as it has a new WS2812fx lib. Maybe that helps.

garfield2412, 2020-10-31:
OK! Installed 3.1.3
GOT IT - Damm Cheap Chinese LED Strips :-) LED 451 was faulty, Cut it out and solderet it together and it works now for the rest of the strip also - but with issues!

First i double checked the hardware wiring and optimized it: 5000uF Capacitor added, wired the strip with + and - from both sides. 100W PS. 

But it seen there is a software issue if you use too many total LED's (>450-460)
Example: if you set static RED Colour with an Value: i.E. 150 - Strip goes perfectly Red - if you change it to 151: Strip changes to a random color.
If you start by RED 0 once again and set it to 120 - perfect Red, if you change the Color by one point, the Color changes to another color. its reproducable with any color, also if you mix colors,
This isue isn't there if you change the Total LED <460 LED.

Strip is also flickering in random colors sometimes, if you use more than 460 LED's in total.

So for the moment i can do a workaround if i start at LED 20 and stop by LED 465, - So i'm loosing 40 LED (20 each side) but it seems to work in this configuration.

garfield2412, 2020-10-31:
Okay... figured out the absolute max for me
482 LED
Starting by No 2 and ending by No 481.
So i have 2 unused LED on the left, and 2 on the right.

It's okay - maybee a memory issue?

bpohvoodoo, 2020-10-31:
I think it is a power issue.

garfield2412, 2020-10-31:
> I think it is a power issue.
> Your strip uses 500 * 0,055mA = 27,5A all LEDs on
> Your power supply has 100W = 20A hopefully at output

I'm sorry... i gave you the wrong Value
.. 60A, 5V, 300W :-)
Seems to be enough.

bpohvoodoo, 2020-11-12:
You can try lowering the Data transmission Speed to 400kHz
by removing the // in front of
//#define LED_TYPE_WS2811