fablab-luenen/McLighting #105: Circuit of the project

larissadj, 2020-11-18:
hi,
can someone help me with the circuit of this project``? I followed step by step the video on youtube. it built the code and all is working properly, but the LED is always off. I used currently the LED LSl-5MIP2W30 5V and the power supply module elego. First, i connected the supply module to the power, then the 5V=VCC; GND = GND, the 3 pins of LEDs have been attached together and connected to PIN 4 of the esp8266. The other side of the LED has also 4 cables,. i didnt connected those one. what is wrong ??please helppp!!!!!!!!!!!!!!!! Thanks

bpohvoodoo, 2020-11-18:
What Esp Module doe you use, what LEDs aure you using. Then send your
results of http://your_device_ip/esp_status.
Maybe you should take some pictures of your circuit and upload them.

larissadj, 2020-11-18:
there is my esp module (see picture), and LED LS-5MIP2W30 [](url)[
[image]
[image]
[image]

](url)

larissadj, 2020-11-18:
[image]

bpohvoodoo, 2020-11-18:
GND of LEDs also has to be connected to GND of the Nodemcu.
https://github.com/toblum/McLighting/wiki/Hardware
I don't see anything. What is connected where?
Did you connect the correct side of the LED Strip to PIN 4? The end with arrows away from Cable? It looks like you connected the wrong end.   DataIN   -> DataOUT
Have you set the definitions.h correctly. If you use DMA (default) you have to connet the leds to pin 3 (RX), no other PIN will work.
Or change to your needs.  
#define USE_WS2812FX_DMA 0
0 = Used PIN is ignored & set to RX/GPIO3;
1 = Used PIN is ignored & set to TX/GPIO1;
2 = Used PIN is ignored & set to D4/GPIO2;
       or comment it out = free Pin config via UI;

larissadj, 2020-11-18:
At the zwo ends of my LED , there is 5pins: 3 together for the port (D0), 1 for GND and 1 for +5V.

For my test, i used only one side of LED.

I used the supply Module voltage like in the picture to connect VCC= 5V and GND = GND

I changed like in the video #define PIN_LED 14 to PIN D4 and i renamed the variable PIN_LED to PIN in all files of my Program.

I conneted the 3 others Pins  of my LED to D4,3v3 and GND. 
 
The esp was connecting to the Computer and the supply voltage to

larissadj, 2020-11-18:
Should i connect the zwo ends of my Controller to the supply voltage?

bpohvoodoo, 2020-11-18:
No matter whatever you insert into PIN_LED is ignored, as long as #define USE_WS2812FX_DMA is not commented out. Please read the comments in text! What video are you talking about? Please see the schematics at https://github.com/toblum/McLighting/wiki/Hardware
The Level shifter is not needed under normal conditions!
Here in Text
5V to red wire LED and NodeMCU Vin
GND to white wire and NodeMCU GND
Green Wire to LED Din
There aren't 5 wires, there are only 3. Red and white are there twice, but are the same.

larissadj, 2020-11-18:
Many thanks , i read again the documentation and try it again. i will give you a Feedback of my results.🙏🏽

larissadj, 2020-11-20:
Hi sorry, i cannot fix the problem. I readed again the file definitions.h and i saw again the schemantics.
i used the only one side of my led. As you described above:

5V to red wire LED and NodeMCU Vin
GND to white wire and NodeMCU GND
Green Wire to LED Din and NodeMCU pin, which is configured!!!!

i didn't commented the #define USE_WS2812FX_DMA. the code is running properly but no effect on the led. (see pictures of my circuit below). thanks

bpohvoodoo, 2020-11-20:
The wireing looks ok.
Please send your results of http://your_device_ip/esp_status.

larissadj, 2020-11-20:
There is the result:
{"HOSTNAME":"larissa","version":"2.2.5","heap":25472,"sketch_size":634512,"free_sketch_space":1462272,"flash_chip_size":4194304,"flash_chip_real_size":4194304,"flash_chip_speed":40000000,"sdk_version":"2.2.2-dev(38a443e)","core_version":"2_7_4","cpu_freq":80,"chip_id":1458415,"animation_lib":"WS2812FX","pin":5,"number_leds":300,"button_mode":"ON","home_assistant":"OFF","legacy_animations":"ON","esp8266_http_updateserver":"ON","state_save":"SPIFFS"}

bpohvoodoo, 2020-11-20:
Ok, now it is getting clearer.
Please update first to my latest version 3.1.3.
Your sketch is version 2.2.5 of toblums original fork.

But inspite of that fact your leds should work now.
The only thing possible is, that you have killed your nodemcu or led strip previosly.
On one picture you connected three white wires to the led strip (D4; 3.3V and GND).
If you ever connected the 3.3V of Nodemcu to 5V of LEDs, then it is very likely that you killed at least one component.

larissadj, 2020-11-20:
@Where did you connect the 5V Power? did you meant the level shifter /power supply ?

bpohvoodoo, 2020-11-21:
https://github.com/fablab-luenen/McLighting/releases
Here is the schema for you.
Beware that if you update to my fork you have to pay attention to the definition.h
like i mentioned before. By default don't connect D4 but RX to LEDs Din

[image]

larissadj, 2020-11-23:
hi, many thanks for your time. It is working now properly bz using the above circuit. I did also a test with the same circuit without the power supply, it also works. Thanks so much. I m so happy thamks!!!!

larissadj, 2020-11-23:
A last question please, how to use node-red to subcribe to the mqtt-topic??. I imported your nodred-file in node-red to get the flow. In the file dfinition.h, i set #define ENABLE_MQTT 0 , to enabe client node. The connection with mqtt was okay. But my result looks like this: 
[image]
May you have please an idea , what is wrong. Thanks

bpohvoodoo, 2020-11-23:
This example has nothing to to with mqtt. It was an Ws proxy to synchronize more than one McLighting. This was implemented in the McLighting Nodes HTML Page. It is not needed anymore and has nothing to do with mqtt.

closed 2020-11-25 by bpohvoodoo