fablab-luenen/McLighting #111: Modify the website

larissadj, 2020-11-25:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

larissadj, 2020-11-25:
Hi,
is it possible to modify the website as my need? I need to add 3 buttons on the website for the 3 colors red, blue and white only. In the file index.htm, i modify the title for example and there is no effect on the website. I need to receive on the terminal red when i click red on the website.  how can i do please to add  this button on the website or whrere can i modify the website`? On the previous version 2.2.5, the modification of the file index.htm was done. But now no effect...Some issues? Thanks

bpohvoodoo, 2020-11-25:
First of all you have to comment out the line
#define USE_HTML_MIN_GZ               // uncomment for using index.htm & edit.htm from PROGMEM instead of SPIFFS
to
//#define USE_HTML_MIN_GZ               // uncomment for using index.htm & edit.htm from PROGMEM instead of SPIFFS
in the definitions.h and the recompile and reupload the firmware.

You can modify the index.htm in the data directory on your own, but you have to upload the files
index.htm
apple-touch-icon.png
edit.htm.gz
favicon.ico.gz
material.woff2
of that folder to the Spiffs.
This is done via:
http://your_device_ip/upload
for each of the mentioned files.

larissadj, 2020-11-26:
My definition.h looks so //#define USE_HTML_MIN_GZ 
i uploaded one by one the above files, because i can choose only one file by upload. At the end, i entered 
http://your_device_ip and my result is the same despite the modification in the index.htm
[image]

closed 2020-11-30 by larissadj