fablab-luenen/McLighting #112: The MQTT ONLINE status does not work as expected

gitpower2017, 2020-12-08:
**Describe the bug**
After Power ON, the MQTT sends an ONLINE once in the field McL/config. A connected client receives this status. A new client or a new query brings an OFFLINE, although the McL is still switched on and working. 

**To Reproduce**
1) MQTT Broker Mosquitto
2) MQTT-client Node-Red
3) MQTT-Client Laptop
4) ESP McL with SK6812

a) Node-Red is connected to the broker. The field McL/config is subscribed.
Power on ESP. Node-Red MQTT says ONLINE
b) Laptop establishes a new connection to the broker The McL/config field is subscribed to. I see an OFFLINE. But the ESP is ONLINE
c) Node-Red MQTT-Client makes a reconnect to Broker. Subscribes to new McL/config. The answer is OFFLINE. But the ESP is ONLINE
d) Power OFF from ESP all MQTT clients get an OFFLINE

bpohvoodoo, 2020-12-08:
Have to check that

bpohvoodoo, 2020-12-15:
Can't reproduce.
Please send your  htp://your_device_ip/esp_status

unbekannt3DE, 2021-01-15:
Works as expected here but i think that the out message for the new status could be sent faster e.g my HASS shows the "old" state for around 5-10 seconds after switching on/off as it waits for the feedback...
[image]

ryancasler, 2021-01-24:
There is a programmable delay in the definitions.h file.  
#define DELAY_MQTT_HA_MESSAGE 5
This is so that it doesn't flood HA with too many messages all at the same time.  It waits for it to be in a stable state and then sends the message.  But it is configurable in the sketch.