fablab-luenen/McLighting #114: McLighting does't work in AP Mode

atomroflman, 2021-01-24:
**Describe the bug**
I have flashed my Wemos D1 mini like described in the installation guide. 
I'd have expected the AP Web UI to offer settings to change the lights settings or a setting to enable the AP Mode in definitions.h.

**To Reproduce**
- Flash controller
- Open Web UI

**Expected behavior**
See description.

**Desktop (please complete the following information):**
 - wemos d1 mini

**Additional context**
definitions.h settings:
```C
//#define USE_WS2812FX_DMA       // LED_PIN is ignored & set to RX/GPIO3  Uses WS2812FX, see: https://github.com/kitesurfer1404/WS2812FX
//#define USE_WS2812FX_UART1     // LED_PIN is ignored & set to D4/GPIO2  Uses WS2812FX, see: https://github.com/kitesurfer1404/WS2812FX
//#define USE_WS2812FX_UART2     // LED_PIN is ignored & set to TX/GPIO1  Uses WS2812FX, see: https://github.com/kitesurfer1404/WS2812FX

// Neopixel
#define LED_PIN 14       // LED_PIN (14 / D5) where neopixel / WS2811 strip is attached
#define NUMLEDS 300       // Number of leds in the strip
//#define LED_TYPE_WS2811  // Uncomment if LED type uses 400 KHz (classic 'v1' (not v2) FLORA pixels, WS2811 drivers)
#define BUILTIN_LED 2    // ESP-12F has the built in LED on GPIO2, see https://github.com/esp8266/Arduino/issues/2192
#define BUTTON 4         // Input pin (4 / D2) for switching the LED strip on / off, connect this PIN to ground to trigger button.

#define HOSTNAME "MyName"   // Friendly hostname

#define HTTP_OTA             // If defined, enable ESP8266HTTPUpdateServer OTA code.
//#define ENABLE_OTA         // If defined, enable Arduino OTA code.
//#define ENABLE_AMQTT         // If defined, enable Async MQTT code, see: https://github.com/marvinroger/async-mqtt-client
//#define ENABLE_MQTT        // If defined, enable MQTT client code, see: https://github.com/toblum/McLighting/wiki/MQTT-API
//#define ENABLE_HOMEASSISTANT // If defined, enable Homeassistant integration, ENABLE_MQTT or ENABLE_AMQTT must be active
#define ENABLE_BUTTON        // If defined, enable button handling code, see: https://github.com/toblum/McLighting/wiki/Button-control
//#define MQTT_HOME_ASSISTANT_SUPPORT // If defined, use AMQTT and select Tools -> IwIP Variant -> Higher Bandwidth
#define ENABLE_LEGACY_ANIMATIONS // Dont disable this for now
//#define ENABLE_E131              // E1.31 implementation
#define USE_HTML_MIN_GZ             //comment for using index.htm & edit.htm from SPIFFs instead of PROGMEM
//#define CUSTOM_WS2812FX_ANIMATIONS  //uncomment and put animations in "custom_ws2812fx_animations.h" 
```

ryancasler, 2021-01-25:
AP mode is used to define your wifi credentials to get the device onto your wifi.  Not to provide an access point for you to log into in order to control the lights.  That wouldn't be very useful if you have to log into the lights every time you wanted to change them.

atomroflman, 2021-01-25:
I'd have expected it to work because in the done todos is written:  'Also enable McLighting in Wifi AP mode.'
I have a light solution where no network access point is available. My mobile connects to the controller in the AP Mode by itself and I'd like to control the lights with my mobile.

bpohvoodoo, 2021-01-25:
Yes, it shoud leave the config mode after 120 seconds of inactivity and is usable then as you expected.

atomroflman, 2021-01-26:
@bpohvoodoo after the config timeout the wifi AP is disabled and I cannot connect anymore.

atomroflman, 2021-01-26:
Okay it worked after a fresh upload of the software. 
Thanks.

closed 2021-01-26 by atomroflman