fablab-luenen/McLighting #117: McLighting V3 "Exception (3)" error

ShakyJojo, 2021-03-18:
Hey, I got a error and i don't get it... I'm using a Wemos D1 mini Pro with McLighting v3, i did everything like in the tutorials and after uploading i'm getting this error:

```
22:19:39.750 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
22:19:39.785 -> 
22:19:39.785 -> Exception (3):
22:19:39.785 -> epc1=0x40100941 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40069368 depc=0x00000000
22:19:39.785 -> 
22:19:39.785 -> >>>stack>>>
22:19:39.785 -> 
22:19:39.785 -> ctx: cont
22:19:39.785 -> sp: 3ffffaa0 end: 3fffffc0 offset: 0190
22:19:39.785 -> 3ffffc30:  00000000 00000027 3ffe8548 402080c0  
22:19:39.785 -> 3ffffc40:  0000dc80 00001b90 00001b90 40100a43  
22:19:39.785 -> 3ffffc50:  3ffe9729 00000000 5357000a 3ffffed8  
22:19:39.785 -> 3ffffc60:  3fff0e1c 3ffe8665 00000020 40100cc7  
22:19:39.818 -> 3ffffc70:  3ffe982b 00000000 00000000 3ffffed8  
22:19:39.818 -> 3ffffc80:  4021fd74 3fff0c54 3fffff20 40236085  
22:19:39.818 -> 3ffffc90:  4021fd74 3fff0c54 3ffe9727 40236070  
22:19:39.818 -> 3ffffca0:  3fff0e1c 00000023 3fff0c54 40211cf6  
22:19:39.818 -> 3ffffcb0:  3fff0e1c 3fff0c54 3ffe8665 40211d45  
22:19:39.818 -> 3ffffcc0:  3fff0e1c 3fff0c54 3ffffec0 4020b95d  
22:19:39.818 -> 3ffffcd0:  00d42c02 000003ec 00002000 00000100  
22:19:39.818 -> 3ffffce0:  00000005 00000020 feefeffe feefeffe  
22:19:39.852 -> 3ffffcf0:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.852 -> 3ffffd00:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.852 -> 3ffffd10:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.852 -> 3ffffd20:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.852 -> 3ffffd30:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.852 -> 3ffffd40:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.852 -> 3ffffd50:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.885 -> 3ffffd60:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.885 -> 3ffffd70:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.885 -> 3ffffd80:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.885 -> 3ffffd90:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.885 -> 3ffffda0:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.885 -> 3ffffdb0:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.885 -> 3ffffdc0:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.918 -> 3ffffdd0:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.918 -> 3ffffde0:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.918 -> 3ffffdf0:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.918 -> 3ffffe00:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.918 -> 3ffffe10:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.918 -> 3ffffe20:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.918 -> 3ffffe30:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.952 -> 3ffffe40:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.952 -> 3ffffe50:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.952 -> 3ffffe60:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.952 -> 3ffffe70:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.952 -> 3ffffe80:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.952 -> 3ffffe90:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.952 -> 3ffffea0:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.987 -> 3ffffeb0:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.987 -> 3ffffec0:  3fff2534 3fff259c 3fff0e1c 40223968  
22:19:39.987 -> 3ffffed0:  feefeffe feefeffe 3fff2500 0010001f  
22:19:39.987 -> 3ffffee0:  80efeffe feefeffe feefeffe feefeffe  
22:19:39.987 -> 3ffffef0:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.987 -> 3fffff00:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.987 -> 3fffff10:  feefeffe feefeffe feefeffe feefeffe  
22:19:39.987 -> 3fffff20:  3ffe8df4 3ffe9850 feefeffe 00000040  
22:19:40.021 -> 3fffff30:  00000001 3ffe9859 feefeffe feefeffe  
22:19:40.021 -> 3fffff40:  feefeffe feefeffe feefeffe feefeffe  
22:19:40.021 -> 3fffff50:  feefeffe feefeffe feefeffe feefeffe  
22:19:40.021 -> 3fffff60:  feefeffe feefeffe feefeffe feefeffe  
22:19:40.021 -> 3fffff70:  feefeffe feefeffe feefeffe feefeffe  
22:19:40.021 -> 3fffff80:  3fff25b4 feefeffe feefeffe feefeffe  
22:19:40.021 -> 3fffff90:  feefeffe feefeffe feefeffe 3fff0da0  
22:19:40.054 -> 3fffffa0:  3fffdad0 00000000 3fff0d60 402228e8  
22:19:40.054 -> 3fffffb0:  feefeffe feefeffe 3ffe87cc 40100f81  
22:19:40.054 -> <<<stack<<<
22:19:40.054 -> 
```
After pasting in ExceptionDecoder:

```
Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
PC: 0x40100941: umm_assimilate_up(uint16_t) at C:\Users\Johannes\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 207
EXCVADDR: 0x40069368

Decoding stack results
0x402080c0: readStateFS() at C:\Users\Johannes\AppData\Local\Temp\arduino_build_429874\sketch/filesystem_functions.h line 158
0x40100a43: umm_free_core(void*) at C:\Users\Johannes\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 351
0x40100cc7: free(void*) at C:\Users\Johannes\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 398
0x4021fd74: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\Johannes\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/HardwareSerial.h line 164
0x40236085: operator delete(void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_op.cc line 48
0x4021fd74: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\Johannes\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/HardwareSerial.h line 164
0x40236070: operator delete[](void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_opv.cc line 33
0x40211cf6: WiFiManagerParameter::setValue(char const*, int) at C:\Users\Johannes\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp line 93
0x40211d45: WiFiManagerParameter::WiFiManagerParameter(char const*, char const*, char const*, int, char const*) at C:\Users\Johannes\Documents\Arduino\libraries\WiFiManager-master\WiFiManager.cpp line 50
0x4020b95d: setup() at C:\Users\Johannes\Documents\Arduino\McLighting/McLighting.ino line 379
0x40223968: fs::FS::_defaultTimeCB() at C:\Users\Johannes\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/FS.h line 243
0x402228e8: loop_wrapper() at C:\Users\Johannes\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 194
```

Maybe you guys can help me? Thanks!

bpohvoodoo, 2021-03-19:
Please reupload the sketch and select to erase all data.

ShakyJojo, 2021-03-19:
I just fixed it by just clean installing all again. Now it works. :D

closed 2021-03-19 by ShakyJojo