fablab-luenen/McLighting #120: error "narrowing conversion of '-1' from 'int' to 'uint8_t' {aka 'unsigned char'} [-Wnarrowing]" during the compilation in Arduino IDE

crexik, 2021-07-08:
Hi, I´m new in Arduino and progrmaing. 
During the compilation in arduino ide stop compilation with error "narrowing conversion of '-1' from 'int' to 'uint8_t' {aka 'unsigned char'} [-Wnarrowing]"

Error refers to line 193 in McLightning.ino where is this command : "uint8_t strIndex[] = {0, -1};"

Original error text : McLighting.ino: In function 'String getValue(String, char, uint8_t)':
McLighting:193:30: error: narrowing conversion of '-1' from 'int' to 'uint8_t' {aka 'unsigned char'} [-Wnarrowing]
  193 |   uint8_t strIndex[] = {0, -1};

Problem is with -1, but i´m not able to solve this problem.

Version of Arduino IDE is 1.8.15

seanasimpson, 2021-07-23:
I'm having this problem as well.

bpohvoodoo, 2021-08-23:
I can not reproduce. Please check the version of your libraries.

closed 2021-08-23 by bpohvoodoo

cralex11, 2022-06-08:
the same, help!